kenzok8/openwrt-packages #263: 还有就是大佬能添加[luci-app-amlogic](https://github.com/ophub/luci-app-amlogic)和 [luci-app-netdata_cn](https://github.com/sirpdboy/luci-app-netdata) 这两个插件吗？

KoWming, 2022-03-24:
[image]
拉取最新的源码编译后[argonneTheme v1.7.2](https://github.com/kenzok78/luci-theme-argonne)主题默认和上图不一样了，比较喜欢上图的样子和配色。

还有就是大佬能添加[luci-app-amlogic](https://github.com/ophub/luci-app-amlogic)和
[luci-app-netdata_cn](https://github.com/sirpdboy/luci-app-netdata)
这两个插件吗？我单独添加一直添加不进去

kenzok8, 2022-03-25:
有人说丑，我就更改过来了，另外如果要很多插件，建议移步至https://github.com/kenzok8/small-package

KoWming, 2022-03-26:
> 有人说丑，我就更改过来了，另外如果要很多插件，建议移步至https://github.com/kenzok8/small-package

是的，众口难调，我用small-package 试试，感谢大佬整理。

closed 2022-03-26 by kenzok8

ieasyseek, 2022-05-01:
> 有人说丑，我就更改过来了，另外如果要很多插件，建议移步至https://github.com/kenzok8/small-package

这个主题不丑啊，相当简洁好看。希望能给个链接。

KoWming, 2022-05-01:
这是来自QQ邮箱的假期自动回复邮件。
 
您好，我最近正在休假中，无法亲自回复您的邮件。我将在假期结束后，尽快给您回复。