kenzok8/openwrt-packages #269: 大佬，编译报错了，怎么解决？

sjkhsl, 2022-04-08:
大佬，编译报错了，怎么解决？

1 out of 1 hunk FAILED -- saving rejects to file go.sum.rej
Patch failed!  Please fix ./patches/010-go-1.18.patch!
make[3]: *** [Makefile:116: /home/sjkhsl/lede/build_dir/target-x86_64_musl/v2ray-core-5.0.3/.prepared_fbb38902d4ef9892bb604f5304bbae84_6664517399ebbbc92a37c5bb081b5c53] Error 1
make[3]: Leaving directory '/home/sjkhsl/lede/feeds/small/v2ray-core'
time: package/feeds/small/v2ray-core/compile#0.13#0.02#0.13
    ERROR: package/feeds/small/v2ray-core failed to build.
make[2]: *** [package/Makefile:116: package/feeds/small/v2ray-core/compile] Error 1
make[2]: Leaving directory '/home/sjkhsl/lede'
make[1]: *** [package/Makefile:110: /home/sjkhsl/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/sjkhsl/lede'
make: *** [/home/sjkhsl/lede/include/toplevel.mk:230: world] Error 2

guochan2019, 2022-04-09:
今天编译，同样的错误，好奇怪

kenzok8, 2022-04-09:
v2ray-core 更新出错，补丁没同时更新

guochan2019, 2022-04-09:
> v2ray-core 更新出错，补丁没同时更新
> 
> rm -rf ./feeds && ./scripts/feeds update -a && ./scripts/feeds install -a && rm -rf ./tmp && make menuconfig

照此解决了，感谢群主

chnhhwz, 2022-04-11:
> 

今天编译出来的固件passwall的v2ray的版本号是无，需要手动把v2ray核心上传解压才能更新组件。马上试试大佬的命令，看看能解决问题不

closed 2022-04-13 by kenzok8

sjkhsl, 2022-04-14:
> > v2ray-core 更新错误，没有同时更新
> > rm -rf ./feeds && ./scripts/feeds update -a && ./scripts/feeds install -a && rm -rf ./tmp && make menuconfig
> 
> 照此解决了，感谢群主

已解决，还有，好像passwall1有问题？又报错，换了passwall2就好了～

kenzok8, 2022-04-14:
这几天上游仓库更新频繁，我也更新频繁，建议不要立刻追新，特别是passwall这几天更新的比较多

chnhhwz, 2022-04-18:
> v2ray-core 更新错误，没有同时更新
>  rm -rf ./feeds &amp;&amp; ./scripts/feeds update -a &amp;&amp; ./scripts/feeds install -a &amp;&amp; rm -rf ./tmp &amp;&amp; make menuconfig

哈，都解决了哈。我很少用QQ邮箱 ，今天设置OMV邮箱通告才来看了一眼