kenzok8/openwrt-packages #274: SSRPLUS x86_64

karafen, 2022-04-17:
Hello, in "Running Mode-global", "Proxy ports-All ports", "Resolve Dns Mode- Use local dns service listen port 5335", apple services do not work (especially push notifications, and sometimes even the appstore itself.
Stubby listening on port 5335 is configured as a dns service

kenzok8, 2022-04-18:
"Running Mode-global" Should use PDNSD TCP query and cache

karafen, 2022-04-18:
> "Running Mode-global" Should use PDNSD TCP query and cache
Thank you very much for your answer, but I apologize in the "Running Mode-global" use PDNSD TCP query and cache" mode that does not open all blocked sites, for example, this error in the dnsmasq logs:
"daemon.info dnsmasq[4623]: 1018 192.168.10.238/58651 forwarded the.hiveos.farm to 217.174.227.102 (this is the provider's dns from resolv.conf.auto)
daemon.info dnsmasq[4623]: 1018 192.168.10.238/58651 reply error is SERVFAIL"
For some reason, the request to this site goes directly to the dns provider, bypassing pdns, and I also noticed that all sites with private cloudflare ip addresses behave similarly
Thank you for your attention.

Dagelinckx, 2022-04-21:
I have more success when I use IP route mode instead of Global

closed 2022-04-23 by kenzok8