kenzok8/openwrt-packages #278: 求教大佬们 编译出错 感谢

Jernin, 2022-04-24:
AILED: lua/CMakeFiles/uloop_lua.dir/uloop.c.o 
/home/jernin/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/bin/x86_64-openwrt-linux-musl-gcc -DJSONC -Duloop_lua_EXPORTS -I/home/jernin/lede/staging_dir/target-x86_64_musl/usr/include/json-c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/jernin/lede/build_dir/target-x86_64_musl/libubox-2021-05-16-b14c4688=libubox-2021-05-16-b14c4688 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/jernin/lede/staging_dir/target-x86_64_musl/usr/include -DNDEBUG -fPIC   -Wall -Werror -Wextra -Werror=implicit-function-declaration -Wformat -Werror=format-security -Werror=format-nonliteral -Os -std=gnu99 -g3 -Wmissing-declarations -Wno-unused-parameter -Os -Wall -Werror --std=gnu99 -g3 -I.. -MD -MT lua/CMakeFiles/uloop_lua.dir/uloop.c.o -MF lua/CMakeFiles/uloop_lua.dir/uloop.c.o.d -o lua/CMakeFiles/uloop_lua.dir/uloop.c.o -c /home/jernin/lede/build_dir/target-x86_64_musl/libubox-2021-05-16-b14c4688/lua/uloop.c
/home/jernin/lede/build_dir/target-x86_64_musl/libubox-2021-05-16-b14c4688/lua/uloop.c:21:10: fatal error: lua.h: No such file or directory
 #include <lua.h>
          ^~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.
make[3]: *** [Makefile:113: /home/jernin/lede/build_dir/target-x86_64_musl/libubox-2021-05-16-b14c4688/.built] Error 1
make[3]: Leaving directory '/home/jernin/lede/package/libs/libubox'
time: package/libs/libubox/compile#3.32#0.70#3.95
    ERROR: package/libs/libubox failed to build.
make[2]: *** [package/Makefile:116: package/libs/libubox/compile] Error 1
make[2]: Leaving directory '/home/jernin/lede'
make[1]: *** [package/Makefile:110: /home/jernin/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/jernin/lede'
make: *** [/home/jernin/lede/include/toplevel.mk:230：world] 错误 2

kenzok8, 2022-04-24:
2021-05-16-b14c4688/lua/uloop.c:21:10: fatal error: lua.h: No such file or directory 目测是这个文件没下载下来（下载失败）
find dl -size -1024c -exec ls -l {} \; && find dl -size -1024c -exec rm -f {} \;
make download

Jernin, 2022-04-24:
> 2021-05-16-b14c4688/lua/uloop.c:21:10: fatal error: lua.h: No such file or directory 目测是这个文件没下载下来（下载失败）
> find dl -size -1024c -exec ls -l {} ; && find dl -size -1024c -exec rm -f {} ;
> make download

大佬 是不是这个命令里｛｝里面是填文件名？  我尝试过删除这个2021-05-16的文件夹重新Make也是不行

kenzok8, 2022-04-25:
复制命令运行即可，这个作用是删除dl下载失败或者0字节的文件（无效文件）

closed 2022-04-25 by kenzok8