kenzok8/openwrt-packages #289: AdGuard Home 一直在运行中，无法关闭

Baojun-Chen, 2022-05-14:
查到了一个问题，原因是 /etc/init.d里面有两个启动文件一个大写一个小写的。删除掉小写的那个 启动文件就正常了。

但是为什么会编译出来两个呢?

[image]

[image]

kenzok8, 2022-05-15:
是Lean源码吗？
因为L大源码net里有AdGuard,所以编译出来etc/init路径下会两个

Baojun-Chen, 2022-05-15:
> 是Lean源码吗？ 因为L大源码net里有AdGuard,所以编译出来etc/init路径下会两个

是有个诶，还取消不了  大佬这种情况该怎么办啊
[image]

ligehhh, 2022-05-18:
后台有进程kill掉就没有了

zombieht, 2022-06-07:
我也遇到这个问题，自带的在config里去掉会报错，没要手动kill好麻烦

kenzok8, 2022-06-08:
我说Lean的源码里有，在net下要删除源码自带的adguardhome
比如 命令  rm -rf feeds/packages/net/adguardhome

closed 2023-04-03 by kenzok8

mcusee, 2024-06-10:
> 除源码自带的adguardhome
> 比如 命令 rm -rf feeds/packages/

删除这个报错

mcusee, 2024-06-10:
> > 除源码自带的adguardhome
> > 比如 命令 rm -rf feeds/packages/
> 
> 删除这个报错
WARNING: Makefile 'package/feeds/kenzo/luci-app-adguardhome/Makefile' has a dependency on 'adguardhome', which does not exist

mcusee, 2024-06-10:
> > > 除源码自带的adguardhome
> > > 比如 命令 rm -rf feeds/packages/
> > 
> > 
> > 删除这个报错
> > WARNING: Makefile 'package/feeds/kenzo/luci-app-adguardhome/Makefile' has a dependency on 'adguardhome', which does not exist

make[3]: *** [Makefile:87: /home/eason/lede/build_dir/target-x86_64_musl/adguardhome-0.107.51/.built] Error 1
make[3]: Leaving directory '/home/eason/lede/feeds/kenzo/adguardhome'
time: package/feeds/kenzo/adguardhome/compile#2.72#4.73#23.79
    ERROR: package/feeds/kenzo/adguardhome failed to build.
make[2]: *** [package/Makefile:116: package/feeds/kenzo/adguardhome/compile] Error 1
make[2]: Leaving directory '/home/eason/lede'
make[1]: *** [package/Makefile:110: /home/eason/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/eason/lede'
make: *** [/home/eason/lede/include/toplevel.mk:231：world] 错误 2