kenzok8/openwrt-packages #291: passwall 规则这段时间都是更新失败，规则主页能打开，是不是规则下架了

adwzlad, 2022-05-19:
2022-05-19 11:42:01: gfwlist 开始更新...
2022-05-19 11:42:15: gfwlist 第1条规则:https://cdn.jsdelivr.net/gh/YW5vbnltb3Vz/domain-list-community@release/gfwlist.txt下载失败！
2022-05-19 11:42:15: gfwlist 第2条规则:https://cdn.jsdelivr.net/gh/Loyalsoldier/v2ray-rules-dat@release/gfw.txt下载失败！
2022-05-19 11:42:15: gfwlist 第3条规则:https://cdn.jsdelivr.net/gh/Loukky/gfwlist-by-loukky/gfwlist.txt下载失败！
2022-05-19 11:42:18: gfwlist 文件下载失败！
2022-05-19 11:42:18: chnroute 开始更新...
2022-05-19 11:42:38: chnroute 第3条规则:https://cdn.jsdelivr.net/gh/soffchen/GeoIP2-CN@release/CN-ip-cidr.txt下载失败！
2022-05-19 11:42:43: chnroute 文件下载失败！
2022-05-19 11:42:43: chnroute6 开始更新...
2022-05-19 11:42:45: chnroute6 更新成功，总规则数 1948 条。
2022-05-19 11:42:45: chnlist 开始更新...
2022-05-19 11:42:48: chnlist 第1条规则:https://cdn.jsdelivr.net/gh/felixonmars/dnsmasq-china-list/accelerated-domains.china.conf下载失败！
2022-05-19 11:42:48: chnlist 第2条规则:https://cdn.jsdelivr.net/gh/felixonmars/dnsmasq-china-list/apple.china.conf下载失败！
2022-05-19 11:42:49: chnlist 第3条规则:https://cdn.jsdelivr.net/gh/felixonmars/dnsmasq-china-list/google.china.conf下载失败！
2022-05-19 11:42:49: chnlist 文件下载失败！

adwzlad, 2022-05-26:
不要开启IPV6

VwEl, 2022-05-30:
替换链接吧。YW5vbnltb3Vz/domain-list-community@release/gfwlist.txt是没有了

Small-tailqwq, 2022-06-04:
应该是jsd被dns污染的缘故，同上替换链接吧

closed 2026-02-24 by kenzok8