kenzok8/openwrt-packages #298: SSRP依赖sagernet-core缺失

wang1zhen, 2022-06-07:
```
Makefile 'package/feeds/kenzo/luci-app-ssr-plus/Makefile' has a dependency on 'sagernet-core', which does not exist
```

wendaoshenyang, 2022-06-07:
同样遇到了，编译报错日志如下：
Configuring luci-i18n-firewall-zh-cn.
Configuring ppp-mod-pppoe.
Configuring luci-i18n-nlbwmon-zh-cn.
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency luci-lib-ipkg for luci-app-ssr-plus
 * pkg_hash_fetch_best_installation_candidate: Packages for luci-app-ssr-plus found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package luci-app-ssr-plus.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-i18n-ssr-plus-zh-cn:
 * 	luci-lib-ipkg
 * opkg_install_cmd: Cannot install package luci-i18n-ssr-plus-zh-cn.
 * check_conflicts_for: The following packages conflict with sagernet-core:
 * check_conflicts_for: 	v2ray-core * check_conflicts_for: 	xray-core * 
 * opkg_install_cmd: Cannot install package sagernet-core.
 * resolve_conffiles: Existing conffile /home/lc/lede/build_dir/target-aarch64_cortex-a53_musl/root-armvirt/etc/config/socat is different from the conffile in the new package. The new conffile will be placed at /home/lc/lede/build_dir/target-aarch64_cortex-a53_musl/root-armvirt/etc/config/socat-opkg.
make[2]: *** [package/Makefile:70: package/install] Error 255
make[2]: Leaving directory '/home/lc/lede'
make[1]: *** [package/Makefile:111: /home/lc/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/lc/lede'
make: *** [/home/lc/lede/include/toplevel.mk:230：world] 错误 2

ligehhh, 2022-06-07:
天灵源码有   https://github.com/xiaorouji/openwrt-passwall/tree/packages/sagernet-core  大佬拉一下

kenzok8, 2022-06-07:
下午就添加了，你们重新拉下

wang1zhen, 2022-06-07:
```
Building targets
../../../../../dl/go-mod-cache/github.com/sagernet/sing@v0.0.0-20220606113732-5efacc1c7db5/common/metadata/addr.go:5:2: package net/netip is not in GOROOT (/home/wang1zhen/openwrt/staging_dir/hostpkg/lib/go-cross/src/net/netip)
```

lisonglin980, 2022-06-07:
Collected errors:
 * check_conflicts_for: The following packages conflict with sagernet-core:
 * check_conflicts_for:         v2ray-core * check_conflicts_for:       xray-core *
 * opkg_install_cmd: Cannot install package luci-app-ssr-plus.
 * check_conflicts_for: The following packages conflict with sagernet-core:
 * check_conflicts_for:         v2ray-core * check_conflicts_for:       xray-core *
 * opkg_install_cmd: Cannot install package sagernet-core.
make[2]: *** [package/Makefile:70: package/install] Error 255
make[2]: Leaving directory '/home/lee/lede'
make[1]: *** [package/Makefile:111: /home/lee/lede/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/lee/lede'
make: *** [/home/lee/lede/include/toplevel.mk:230: world] Error 2

ligehhh, 2022-06-07:
> Collected errors:
>
> * check_conflicts_for: The following packages conflict with sagernet-core:
> * check_conflicts_for:         v2ray-core * check_conflicts_for:       xray-core *
> * opkg_install_cmd: [...]

取消一个 现在lede的源码还有问题

wendaoshenyang, 2022-06-08:
同楼上：
Collected errors:
 * check_conflicts_for: The following packages conflict with sagernet-core:
 * check_conflicts_for: 	v2ray-core * check_conflicts_for: 	xray-core * 
 * opkg_install_cmd: Cannot install package luci-app-ssr-plus.
 * check_conflicts_for: The following packages conflict with sagernet-core:
 * check_conflicts_for: 	v2ray-core * check_conflicts_for: 	xray-core * 
 * opkg_install_cmd: Cannot install package sagernet-core.
 * resolve_conffiles: Existing conffile /home/lc/lede/build_dir/target-aarch64_cortex-a53_musl/root-armvirt/etc/config/socat is different from the conffile in the new package. The new conffile will be placed at /home/lc/lede/build_dir/target-aarch64_cortex-a53_musl/root-armvirt/etc/config/socat-opkg.
make[2]: *** [package/Makefile:70: package/install] Error 255
make[2]: Leaving directory '/home/lc/lede'
make[1]: *** [package/Makefile:111: /home/lc/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/lc/lede'
make: *** [/home/lc/lede/include/toplevel.mk:230：world] 错误 2

xztxy, 2022-06-08:
为什么我重新拉了还是一样的问题啊

kenzok8, 2022-06-08:
如果是写入feeds的
rm -rf ./feeds && ./scripts/feeds update -a && ./scripts/feeds install -a && rm -rf ./tmp && make menuconfig
如果是git到lean/package目录下的
cd lean/package 运行rm -rf 该文件夹

wang1zhen, 2022-06-08:
可以了，顺便问一下，解锁网易云的插件依赖ucode，可以也加进来吗？

xztxy, 2022-06-08:
> 

好的，我去试试！感谢！

xztxy, 2022-06-08:
> 如果是写入feeds的 rm -rf ./feeds && ./scripts/feeds update -a && ./scripts/feeds install -a && rm -rf ./tmp && make menuconfig 如果是git到lean/package目录下的 cd lean/package 运行rm -rf 该文件夹
好的，我去试试！感谢！

xztxy, 2022-06-08:
> 如果是写入feeds的 rm -rf ./feeds && ./scripts/feeds update -a && ./scripts/feeds install -a && rm -rf ./tmp && make menuconfig 如果是git到lean/package目录下的 cd lean/package 运行rm -rf 该文件夹

好的，我去试试，感谢！

zlfzy, 2022-06-08:
> 如果是写入feeds的 rm -rf ./feeds && ./scripts/feeds update -a && ./scripts/feeds install -a && rm -rf ./tmp && make menuconfig 如果是git到lean/package目录下的 cd lean/package 运行rm -rf 该文件夹

一样不行
```Collected errors:
 * check_conflicts_for: The following packages conflict with sagernet-core:
 * check_conflicts_for:         v2ray-core * check_conflicts_for:       xray-core * 
 * opkg_install_cmd: Cannot install package luci-app-ssr-plus.
 * check_conflicts_for: The following packages conflict with sagernet-core:
 * check_conflicts_for:         v2ray-core * check_conflicts_for:       xray-core * 
 * opkg_install_cmd: Cannot install package sagernet-core.
make[2]: *** [package/Makefile:70: package/install] Error 255
make[2]: Leaving directory '/home/test/lede'
make[1]: *** [package/Makefile:111: /home/test/lede/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/test/lede'
make: *** [/home/test/lede/include/toplevel.mk:230: world] Error 2```

WYC-2020, 2022-06-08:
这明显就是冲突嘛，不要选v2ray-core

wang1zhen, 2022-06-08:
```
Building targets
../../../../../dl/go-mod-cache/github.com/sagernet/sing@v0.0.0-20220606113732-5efacc1c7db5/common/metadata/addr.go:5:2: package net/netip is not in GOROOT (/home/wang1zhen/openwrt/staging_dir/hostpkg/lib/go-cross/src/net/netip)

make[3]: *** [Makefile:88: /home/wang1zhen/openwrt/build_dir/target-x86_64_musl/sagernet-core-5.0.12-2022-06-07-5590c5fe/.built] Error 1
```

wang1zhen, 2022-06-09:
```
Building targets
../../../../../dl/go-mod-cache/github.com/sagernet/sing@v0.0.0-20220606113732-5efacc1c7db5/common/metadata/addr.go:5:2: package net/netip is not in GOROOT (/home/wang1zhen/openwrt/staging_dir/hostpkg/lib/go-cross/src/net/netip)
```

是sagernet-core的makefile有问题吗？

WYC-2020, 2022-06-09:
go 1.18

wang1zhen, 2022-06-09:
你好，问一下该怎么解决？

zheyiyuezhuo, 2022-06-11:
我也好像这个报错
```sh
Collected errors:
 * check_conflicts_for: The following packages conflict with sagernet-core:
 * check_conflicts_for:         v2ray-core * check_conflicts_for:       xray-core *
 * opkg_install_cmd: Cannot install package luci-app-ssr-plus.
 * check_conflicts_for: The following packages conflict with sagernet-core:
 * check_conflicts_for:         v2ray-core * check_conflicts_for:       xray-core *
 * opkg_install_cmd: Cannot install package sagernet-core.
 * resolve_conffiles: Existing conffile /home/sic/lede/build_dir/target-x86_64_musl/root-x86/etc/config/socat is different from the conffile in the new package. The new conffile will be placed at /home/sic/lede/build_dir/target-x86_64_musl/root-x86/etc/config/socat-opkg.
make[2]: *** [package/Makefile:70: package/install] Error 255
make[2]: Leaving directory '/home/sic/lede'
make[1]: *** [package/Makefile:111: /home/sic/lede/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/sic/lede'
make: *** [/home/sic/lede/include/toplevel.mk:230: world] Error 2
```

然后取消了ssr-plus的sagernet-core这项就编译成功了，就是不知道ssr-plus会有什么问题

Gzxhwq, 2022-06-13:
SSRP切换到Sagernet内核了，而且把Sagernet内核和V2ray/Xray内核做了互斥处理。
Passwall现在还是用V2ray/Xray，所以Passwall和SSRP也成了互斥了。
想要同时编译SSRP和Passwall的话，要把SSRP的Sagernet-core去掉，这样就会让SSRP使用Xray-core了

wang1zhen, 2022-06-13:
> SSRP切换到Sagernet内核了，而且把Sagernet内核和V2ray/Xray内核做了互斥处理。 Passwall现在还是用V2ray/Xray，所以Passwall和SSRP也成了互斥了。 想要同时编译SSRP和Passwall的话，要把SSRP的Sagernet-core去掉，这样就会让SSRP使用Xray-core了

make menuconfig时取消sagernet-core进行编译，仍然会有报错：`package net/netip is not in GOROOT`，`ERROR: package/feeds/kenzo/sagernet-core failed to build.`

Gzxhwq, 2022-06-13:
> make menuconfig时取消sagernet-core进行编译，仍然会有报错：`package net/netip is not in GOROOT`，`ERROR: package/feeds/kenzo/sagernet-core failed to build.`

你只是取消了SSRP的那个选项，并没有取消Sagernet-core这个包，你用"/"搜一下这个包的位置，然后手动取消

wang1zhen, 2022-06-13:
> 你只是取消了SSRP的那个选项，并没有取消Sagernet-core这个包，你用"/"搜一下这个包的位置，然后手动取消

感谢，编译成功了。另外问一下，在取消了passwall, passwall2并且勾选了sagernet-core的基础上，编译就会产生上面的`package net/netip is not in GOROOT`，是否与golang的编译环境有关？无论我是否安装`golang-go`与否，都会得到相同的报错。

closed 2022-06-26 by kenzok8