kenzok8/openwrt-packages #299: lean 源码编译插件没覆盖

zhangguanzhang, 2022-06-09:
ref: https://github.com/kenzok8/openwrt-packages/issues/103

```
$ find feeds -type d -name 'luci-app-*' | awk -F/ '{if(a[$NF]){printf "%s/ %s\n",$0,a[$NF]}else{a[$NF]=$0}}' 
feeds/others/luci-app-aliyundrive-webdav/ feeds/luci/applications/luci-app-aliyundrive-webdav
feeds/others/luci-app-diskman/ feeds/luci/applications/luci-app-diskman
feeds/others/luci-app-dockerman/ feeds/luci/applications/luci-app-dockerman
feeds/others/luci-app-easymesh/ feeds/luci/applications/luci-app-easymesh
feeds/others/luci-app-pushbot/ feeds/luci/applications/luci-app-pushbot
feeds/others/luci-app-serverchan/ feeds/luci/applications/luci-app-serverchan
```
这是 luci-app的，没有被覆盖，因为我看 lean 的 aliyunwebdev是 1.3.2 ，你的是 1.5.0 编译出来的是 ipk 是 1.3.2，我 shell 处理了上面几个目录
```
find feeds -type d -name 'luci-app-*' | awk -F/ '{if(a[$NF]){printf "%s/ %s\n",$0,a[$NF]}else{a[$NF]=$0}}' | xargs -r -n2 rsync -av --delete
```
然后编译发现luci-app是1.5.0的，但是核心还是1.3.2的
```
feeds/others/aliyundrive-webdav/ feeds/packages/multimedia/aliyundrive-webdav
```

zhangguanzhang, 2022-06-09:
我自己暂时写了个复杂的来处理。。。，不知道有更好的办法不
```bash
        cd openwrt
        # 添加的 feeds 应用包优先于自带的 feed 里的 app
        echo "重复的包检测：👇"
        ./scripts/feeds list  | awk '{if(a[$1]){print $1}else{a[$1]++}}'
        echo "重复的包检测：👆"
        ./scripts/feeds list  | awk '{if(a[$1]){print $1}else{a[$1]++}}' | while read pkg_name;do
            # 目录是 / 分隔，feeds/xxx/ 一样就不打印
            find feeds/ -maxdepth 4 -type d -name $pkg_name | \
              awk -F/ 'NR==1{a[$2]=$0};NR==2{if(!a[$2]){for(i in a){if(a[i]){printf "%s/ %s\n",$0,a[i]}}}}' | \
              xargs -r -n2 echo  👉 rsync -av --delete
            find feeds/ -maxdepth 4 -type d -name $pkg_name | \
              awk -F/ 'NR==1{a[$2]=$0};NR==2{if(!a[$2]){for(i in a){if(a[i]){printf "%s/ %s\n",$0,a[i]}}}}' | \
              xargs -r -n2 rsync -av --delete
        done
```

canmengxian, 2022-08-16:
添加你要的feeds源，为第一行

Actuallymax, 2022-08-16:
> 添加你要的feeds源，为第一行

没有用，编译出来的还是lean的老版本

canmengxian, 2022-08-21:
> > 添加你要的feeds源，为第一行
> 
> 没有用，编译出来的还是lean的老版本

有没有 clean

jkingben, 2023-04-04:
aliyunwebdev在l大packages的multimedia里面，用现在的替换掉旧的，自己写个脚本维护下就行

closed 2023-05-12 by kenzok8