kenzok8/openwrt-packages #306: lede源码，使用本仓库编译mosdns启动不了。

jaydiao521, 2022-06-25:
/usr/bin/mosdns -dir /etc/mosdns/ -c def_config.yaml
2022-06-25T07:14:02.852Z        info    v3/main.go:157  current working directory: /etc/mosdns/
2022-06-25T07:14:02.853Z        info    v3/main.go:223  mosdns ver: v3.9.0
2022-06-25T07:14:02.853Z        info    v3/main.go:224  arch: amd64, os: linux, go: go1.18.3
2022-06-25T07:14:02.853Z        info    coremain/run.go:65      loading config  {"file": "def_config.yaml"}
2022-06-25T07:14:02.854Z        fatal   coremain/run.go:37      failed to load config   {"error": "failed to parse the config file: yaml: unmarshal errors:\n  line 7: field data_providers not found in type coremain.Config\n  line 41: field plugins not found in type coremain.Config\n  line 161: field servers not found in type coremain.Config"}
以前用的很好。

jaydiao521, 2022-06-25:
看了下原作者的库，应该是原作者加了新的分支，而大佬的库的 V4用了3.9的内核导致mosdns启动不了，建议重新设置一下。很好的插件。

lisonglin980, 2022-06-25:
luci-mosdns是1.4的只支持mosdns coreV4,而你编译的core是V3.9。V4是测试版

kenzok8, 2022-06-25:
同步上游

closed 2022-06-26 by kenzok8