kenzok8/openwrt-packages #308: 编译最新的mosdns报错

jinjin327, 2022-06-26:
WARNING: Makefile 'package/feeds/small8/luci-app-mosdns/Makefile' has a dependency on 'mosdns_neo', which does not exist

ligehhh, 2022-06-26:
缺依赖了

ligehhh, 2022-06-26:
https://github.com/xcf13363175/openwrt-mos/blob/master/mosdns_neo/Makefile   这个应该是依赖包 大佬添加一下

kenzok8, 2022-06-26:
修复了

aiemassfiria, 2022-06-26:
Still error

jinjin327, 2022-06-27:
> 修复了

还是提示缺少依赖mosdns_neo

zhangguanzhang, 2022-06-27:
还是提示缺少依赖mosdns_neo, +1

kenzok8, 2022-06-27:
rm -rf feeds/packages/net/mosdns

zhangguanzhang, 2022-06-27:
> rm -rf feeds/packages/net/mosdns
> 
> 把L大的源码自带的mosdns删除

不行，可以看 
https://github.com/zhangguanzhang/Actions-OpenWrt/runs/7062937300?check_suite_focus=true#step:9:458
https://github.com/zhangguanzhang/Actions-OpenWrt/runs/7062937300?check_suite_focus=true#step:15:77

WYC-2020, 2022-06-27:
这玩意吧那个mosdns_neo改成mosdns

lisonglin980, 2022-06-27:
BuildPackage 变量
PKG_*定义了软件包相关的一些基本信息，如下所示：
PKG_NAME -包的名称,可以通过menuconfig和ipkg看到。 不要使用下划线。

mosdns_neo应该改成mosdns-neo吧

lisonglin980, 2022-06-27:
> rm -rf feeds/packages/net/mosdns

rm了一样报错找不到依赖mosdns_neo

WYC-2020, 2022-06-27:
> BuildPackage 变量 PKG_*定义了软件包相关的一些基本信息，如下所示： PKG_NAME -包的名称,可以通过menuconfig和ipkg看到。 不要使用下划线。
> 
> mosdns_neo应该改成mosdns-neo吧

看上游嘛，都给你说了吧那个依赖去掉neo，这就是每个人定义的包名不同而已，l大不带neo，这个插件的作者也去掉neo了，这库应该是作者以前自己私自明名的而已

lisonglin980, 2022-06-27:
> > BuildPackage 变量 PKG_*定义了软件包相关的一些基本信息，如下所示： PKG_NAME -包的名称,可以通过menuconfig和ipkg看到。 不要使用下划线。
> > mosdns_neo应该改成mosdns-neo吧
> 
> 看上游嘛，都给你说了吧那个依赖去掉neo，这就是每个人定义的包名不同而已，l大不带neo，这个插件的作者也去掉neo了，这库应该是作者以前自己私自明名的而已

直接把\feeds\kenzo\luci-app-mosdns\Makefile里面的LUCI_DEPENDS:=+mosdns_neo这里删掉_neo吗

zhangguanzhang, 2022-06-27:
> rm -rf feeds/packages/net/mosdns
> 
> 把L大的源码自带的mosdns删除

https://github.com/kenzok8/openwrt-packages/blob/9b57ea758278ad94326c088725e06f1baa3eade2/luci-app-mosdns/Makefile#L9 这里你没改

lisonglin980, 2022-06-27:
> > rm -rf feeds/packages/net/mosdns
> > 把L大的源码自带的mosdns删除
> 
> https://github.com/kenzok8/openwrt-packages/blob/9b57ea758278ad94326c088725e06f1baa3eade2/luci-app-mosdns/Makefile#L9
> 
> 这里你没改

我改的就是这个地方呀，只有这一处吗

zhangguanzhang, 2022-06-28:
> > > rm -rf feeds/packages/net/mosdns
> > > 把L大的源码自带的mosdns删除
> > 
> > 
> > https://github.com/kenzok8/openwrt-packages/blob/9b57ea758278ad94326c088725e06f1baa3eade2/luci-app-mosdns/Makefile#L9
> > 
> > 这里你没改
> 
> 我改的就是这个地方呀，只有这一处吗
```
find -type f -name Makefile -exec grep -l mosdns_neo {} \;
```
命令查找下看看

wx1wxc, 2022-06-29:
` * pkg_hash_check_unresolved: cannot find dependency mosdns-neo for luci-app-mosdns
 * pkg_hash_fetch_best_installation_candidate: Packages for luci-app-mosdns found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package luci-app-mosdns.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-i18n-mosdns-zh-cn:
 * 	mosdns-neo
 * opkg_install_cmd: Cannot install package luci-i18n-mosdns-zh-cn.
make[2]: *** [package/Makefile:70: package/install] Error 255`

今天编译还是出错

lee29, 2022-06-29:
> 

云编译yml可以这样改
`    - name: 去除重复应用包
      run: |
         cd openwrt

         sudo rm -rf ./feeds/packages/net/mosdns && cp -rf ./feeds/kenzo/mosdns ./feeds/packages/net/mosdns
         sed -i 's/mosdns_neo/mosdns/g' ./feeds/kenzo/luci-app-mosdns/Makefile
         sed -i 's/mosdns_neo/mosdns/g' ./feeds/kenzo/mosdns/Makefile
         sed -i 's/mosdns_neo/mosdns/g' ./feeds/packages/net/mosdns/Makefile
 
         
               
    - name: 安装 feeds 应用包
      run: cd openwrt && ./scripts/feeds install -a`

jonwhoo, 2022-07-05:
这样处理不报错了，编译一下看看

zhangguanzhang, 2022-07-06:
@kenzok8 老大为啥不改下更新下

wangwei123321, 2022-07-06:
问题解决了么，怎么解决啊

kenzok8, 2022-07-13:
抱歉，最近搬家，所以耽搁了修复了

closed 2023-04-03 by kenzok8