kenzok8/openwrt-packages #320: adguardhome编译后有两个启动项 一个大写 一个小写

tiantaichenhui, 2022-08-12:
是lean的源码 已经删除feeds/packages/net/adguardhome文件夹
但是编译出来还是有2个adguardhome启动项 如果不禁用小写那个 那么adguardhome运行会异常 
[image]

Marionic0723, 2022-08-17:
就是说，莫名其妙的丢配置文件（明明设置工作路径为etc，结果重启完就没了，一看全在tmp），或者造成重启啥的两分钟都重启不了。
https://github.com/kenzok8/openwrt-packages/issues/265 这也有人讨论过了，貌似要把小写的那个删掉？我用的是原版openwrt，就说嘛这玩意bug咋这么多。

tiantaichenhui, 2022-08-17:
> 就是说它的妙用丢文件（明明设置工作路径等，结果重启也有）名，一看全在tmp，或者造成的两两重启都重启不了。 #265有人讨论过了，我用的是原版openwrt，就嘛这玩意儿咋这么多。

是的 好多人碰到这个问题。搜索Issues 好多人碰到一样的问题，目测有一年多了  但一直没有解决这个问题 。

mahakalaya, 2023-03-18:
删除小写那个

kenzok8, 2023-03-20:
如果是L大的源码找到
lede/package/lean/default-settings/files/zzz-default-settings
添加一句
chmod 0755 /etc/init.d/*
如果是Lienol的源码，他本来就添加了
还是权限的问题

jkingben, 2023-03-24:
feed update命令后删除掉原固件里的多余配置
sed -i '/files\/adguardhome.*/d' xxxx/packages/net/adguardhome/Makefile
sed -i '/$(INSTALL_DIR) $(1)\/etc/d' xxxx/packages/net/adguardhome/Makefile

lean固件的adguardhome不带UI，只是个简单的配置，有luci情况下已经不需要那几个会产生冲突的文件，以luci的为主。

kenzok8, 2023-03-25:
> feed update命令后删除掉原固件里的多余配置 sed -i '/files/adguardhome.*/d' xxxx/packages/net/adguardhome/Makefile sed -i '/$(INSTALL_DIR) $(1)/etc/d' xxxx/packages/net/adguardhome/Makefile
> 
> lean固件的adguardhome不带UI，只是个简单的配置，有luci情况下已经不需要那几个会产生冲突的文件，以luci的为主。

楼上你的命令存在问题的，如果你要删除./files/adguardhome.init  与./files/adguardhome.config  
如果达到的效果是
$(INSTALL_BIN) $(1)/etc/init.d/adguardhome
$(INSTALL_DATA) $(1)/etc/config/adguardhome
那么命令不是这样写的
正确的是
sed -i -e 's/\.\/files\/adguardhome\.init //g' -e 's/\.\/files\/adguardhome\.config //g' adguardhome/Makefile

kenzok8, 2023-03-25:
楼上你的命令存在问题的，如果你要删除./files/adguardhome.init  与./files/adguardhome.config  
如果达到的效果是
'$(INSTALL_DATA) $(1)/etc/config/adguardhome'
'$(INSTALL_DATA)  $(1)/etc/config/adguardhome'
那么命令不是这样写的
正确的是
sed -i -e 's/\.\/files\/adguardhome\.init //g' -e 's/\.\/files\/adguardhome\.config //g' xxxx/packages/net/adguardhome/Makefile

jkingben, 2023-03-25:
没啥问题，提取的是里面的关键字，其他地方不含有这些关键字。我已经编译过了

jkingben, 2023-03-25:
[image]
就是把这4条删除掉

kenzok8, 2023-03-25:
那你还不如直接做个补丁更稳定

kenzok8, 2023-03-25:
那最后一个-是删除空行

tonyzzt001, 2024-04-14:
> 那你还不如直接做个补丁更稳定 [adguardhome.patch](https://github.com/kenzok8/openwrt-packages/files/11068672/adguardhome.patch) patch -p1 < adguardhome.patch

你好请问怎么使用这个补丁呢？我执行后显示makefile不匹配……谢谢！

tonyzzt001, 2024-04-14:
> 楼上你的命令存在问题的，如果你要删除./files/adguardhome.init 与./files/adguardhome.config 如果达到的效果是 '$(INSTALL_DATA) $(1)/etc/config/adguardhome' '$(INSTALL_DATA) $(1)/etc/config/adguardhome' 那么命令不是这样写的 正确的是 sed -i -e [...]

你好，这个命令对现在的lede版本还有用吗？