kenzok8/openwrt-packages #322: 报没找到'lua-neturl'

Racon94, 2022-08-14:
用的最新版lede
```bash
...
WARNING: No feed for package 'lua-neturl' found
...
WARNING: Makefile 'package/feeds/kenzo/luci-app-ssr-plus/Makefile' has a dependency on 'lua-neturl', which does not exist
...
```

Actuallymax, 2022-08-14:
一样的问题

Racon94, 2022-08-14:
这是来自QQ邮箱的自动回复邮件。已收到。

S0MEW0LF, 2022-08-14:
同样遇到这个问题。

kenzok8, 2022-08-14:
fw876上游库增加了插件，已同步

WYC-2020, 2022-08-14:
包名字错误而已，自己改

kontorol, 2022-08-14:
Any solution ??

kontorol, 2022-08-14:
> 

Where to change ?

Racon94, 2022-09-03:
今天又出现了，缺少lua-neturl
```
WARNING: Makefile 'package/feeds/kenzo/luci-app-bypass/Makefile' has a dependency on 'lua-neturl', which does not exist
WARNING: Makefile 'package/feeds/kenzo/luci-app-filebrowser/Makefile' has a dependency on 'filebrowser', which does not exist
WARNING: Makefile 'package/feeds/kenzo/luci-app-ssr-plus/Makefile' has a dependency on 'lua-neturl', which does not exist
```

Racon94, 2022-09-03:
自己动手丰衣足食，直接fork这个仓库，把隔壁small-package的lua-neturl文件夹复制进去就好了。记得引用自己的库

closed 2022-09-03 by Racon94