kenzok8/openwrt-packages #328: xray-core出问题了？

sjkhsl, 2022-08-30:
编译时xray-core报错了！

```
/home/sjkhsl/lede/staging_dir/target-x86_64_musl/usr/share/gocode/src does not exist, skipping symlinks

touch /home/sjkhsl/lede/build_dir/target-x86_64_musl/Xray-core-1.5.9/.configured_1cd727ac7b9ebbe4ee453b5ddb8504b3
rm -f /home/sjkhsl/lede/build_dir/target-x86_64_musl/Xray-core-1.5.9/.built
touch /home/sjkhsl/lede/build_dir/target-x86_64_musl/Xray-core-1.5.9/.built_check
CONFIG_GOLANG_MOD_CACHE_WORLD_READABLE="" GO_BUILD_CACHE_DIR="/home/sjkhsl/lede/tmp/go-build" GO_MOD_CACHE_DIR="/home/sjkhsl/lede/dl/go-mod-cache" GO_MOD_ARGS="-modcacherw" GO_PKG="github.com/xtls/xray-core" GO_INSTALL_EXTRA="" GO_INSTALL_ALL="" GO_SOURCE_ONLY="" GO_BUILD_PKG="github.com/xtls/xray-core/main" GO_EXCLUDES="" GO_GO_GENERATE="" GO_INSTALL_BIN_PATH="/usr/bin" BUILD_DIR="/home/sjkhsl/lede/build_dir/target-x86_64_musl/Xray-core-1.5.9" GO_BUILD_DIR="/home/sjkhsl/lede/build_dir/target-x86_64_musl/Xray-core-1.5.9/.go_work/build" GO_BUILD_BIN_DIR="/home/sjkhsl/lede/build_dir/target-x86_64_musl/Xray-core-1.5.9/.go_work/build/bin" GO_BUILD_DEPENDS_PATH="/usr/share/gocode" GO_BUILD_DEPENDS_SRC="/home/sjkhsl/lede/staging_dir/target-x86_64_musl/usr/share/gocode/src" GOOS="linux" GOARCH="amd64" GO386="" GOAMD64="v1" GOARM="" GOMIPS="" GOMIPS64="" GOPPC64="" CGO_ENABLED=1 CC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" CGO_CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/sjkhsl/lede/build_dir/target-x86_64_musl/Xray-core-1.5.9=Xray-core-1.5.9 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_CPPFLAGS="-I/home/sjkhsl/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include -I/home/sjkhsl/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify -I/home/sjkhsl/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include" CGO_CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/sjkhsl/lede/build_dir/target-x86_64_musl/Xray-core-1.5.9=Xray-core-1.5.9 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_LDFLAGS="-L/home/sjkhsl/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/home/sjkhsl/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -znow -zrelro" GOPATH="/home/sjkhsl/lede/build_dir/target-x86_64_musl/Xray-core-1.5.9/.go_work/build" GOCACHE="/home/sjkhsl/lede/tmp/go-build" GOMODCACHE="/home/sjkhsl/lede/dl/go-mod-cache" GOENV=off /usr/bin/env bash /home/sjkhsl/lede/feeds/packages/lang/golang//golang-build.sh build -v -buildvcs=false -trimpath -ldflags "all=-buildid '1661851146' -linkmode external -extldflags '-L/home/sjkhsl/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/home/sjkhsl/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -Wl,-z,now -Wl,-z,relro'"        -ldflags "-s -w -X github.com/xtls/xray-core/core.build=OpenWrt -X github.com/xtls/xray-core/core.version=1.5.9 -buildid '1661851146' -linkmode external -extldflags '-L/home/sjkhsl/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/home/sjkhsl/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -Wl,-z,now -Wl,-z,relro'"  
Finding targets

Building targets
../../../../../dl/go-mod-cache/github.com/sagernet/sing-shadowsocks@v0.0.0-20220716012931-952ae62e05d7/shadowaead_2022/protocol.go:28:2: no required module provides package github.com/sagernet/sing/common/random; to add it:
        go get github.com/sagernet/sing/common/random
../../../../../dl/go-mod-cache/github.com/sagernet/sing-shadowsocks@v0.0.0-20220716012931-952ae62e05d7/shadowaead_2022/service.go:28:2: no required module provides package github.com/sagernet/sing/common/replay; to add it:
        go get github.com/sagernet/sing/common/replay
../../../../../dl/go-mod-cache/github.com/sagernet/sing@v0.0.0-20220714145306-09b55ce4b6d0/common/bufio/conn.go:15:2: no required module provides package github.com/sagernet/sing/common/task; to add it:
        go get github.com/sagernet/sing/common/task

make[3]: *** [Makefile:51: /home/sjkhsl/lede/build_dir/target-x86_64_musl/Xray-core-1.5.9/.built] Error 1
make[3]: Leaving directory '/home/sjkhsl/lede/feeds/small/xray-core'
time: package/feeds/small/xray-core/compile#1.57#1.28#1.75
    ERROR: package/feeds/small/xray-core failed to build.
make[2]: *** [package/Makefile:116: package/feeds/small/xray-core/compile] Error 1
make[2]: Leaving directory '/home/sjkhsl/lede'
make[1]: *** [package/Makefile:110: /home/sjkhsl/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/sjkhsl/lede'
make: *** [/home/sjkhsl/lede/include/toplevel.mk:230: world] Error 2
```

dukmonki, 2022-09-01:
怎么回事？

haruhitamako, 2022-09-02:
一样遇到了

dukmonki, 2022-09-04:
有人来说说么？

sjkhsl, 2022-09-04:
> 有人来说说么？

大佬不知道忙啥呢，感觉好久不露头了～

closed 2023-01-30 by kenzok8