kenzok8/openwrt-packages #330: 家人们，上个星期还没问题，今天突然就报错了，求帮助

Baojun-Chen, 2022-09-02:
Package trojan-plus is missing dependencies for the following libraries:
libboost_program_options.so.1.79.0
libboost_system.so.1.79.0
make[3]: *** [Makefile:78: /home/cbj/lede/bin/packages/x86_64/small/trojan-plus_10.0.3-2_x86_64.ipk] Error 1
make[3]: Leaving directory '/home/cbj/lede/feeds/small/trojan-plus'
time: package/feeds/small/trojan-plus/compile#0.16#0.07#0.20
    ERROR: package/feeds/small/trojan-plus failed to build.
make[2]: *** [package/Makefile:116: package/feeds/small/trojan-plus/compile] Error 1
make[2]: Leaving directory '/home/cbj/lede'
make[1]: *** [package/Makefile:110: /home/cbj/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/cbj/lede'
make: *** [/home/cbj/lede/include/toplevel.mk:230：world] 错误 2

Racon94, 2022-09-03:
make clean

kenzok8, 2022-09-05:
rm -rf ./feeds && ./scripts/feeds update -a && ./scripts/feeds install -a && rm -rf ./tmp && make menuconfig

closed 2022-10-04 by kenzok8