kenzok8/openwrt-packages #331: which does not exist configuration written to .config

seaks, 2022-09-03:
WARNING: Makefile 'package/feeds/kenzo/luci-app-bypass/Makefile' has a dependency on 'lua-neturl', which does not exist
WARNING: Makefile 'package/feeds/kenzo/luci-app-filebrowser/Makefile' has a dependency on 'filebrowser', which does not exist
WARNING: Makefile 'package/feeds/kenzo/luci-app-ssr-plus/Makefile' has a dependency on 'lua-neturl', which does not exist
configuration written to .config

Ginua8292, 2022-09-03:
昨天还好好的!今天就少了依赖！

X-Firefly, 2022-09-03:
同样提示依赖错误

seaks, 2022-09-03:
> 昨天还好好的!今天就少了依赖！

> 同样提示依赖错误

clean后我拉这个库 就好了 sed -i '$a src-git openwrtpasswall https://github.com/xiaorouji/openwrt-passwall' feeds.conf.default

Racon94, 2022-09-04:
我昨天提过，应该是kenzo误删了吧？我是自己fork了一个库，自己加进去了。也可以像楼上说的，加别人的feeds库
https://github.com/kenzok8/openwrt-packages/issues/322#issue-1338104317

Kisoul, 2022-09-04:
我也出现了这个问题

closed 2022-10-04 by kenzok8