kenzok8/openwrt-packages #333: 今天编译又报错

X-Firefly, 2022-09-13:
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency samba4-libs for samba4-server
 * pkg_hash_fetch_best_installation_candidate: Packages for samba4-server found, but incompatible with the architectures configured
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for autosamba:
 * 	samba4-libs
 * opkg_install_cmd: Cannot install package autosamba.
 * check_conflicts_for: The following packages conflict with samba4-server:
 * check_conflicts_for: 	samba36-server * 
 * opkg_install_cmd: Cannot install package samba4-server.
make[2]: *** [package/Makefile:70: package/install] Error 255

PaPaB1, 2022-09-16:
跟楼上一样，遇到了同样的问题，是不是 samba4的依赖库不对？

Jieoz, 2022-09-19:
---
Configuring luci-i18n-nlbwmon-zh-cn.
 * check_conflicts_for: The following packages conflict with kmod-fast-classifier:
 * check_conflicts_for: 	kmod-shortcut-fe-cm * 
 * opkg_install_cmd: Cannot install package kmod-fast-classifier.
 * pkg_hash_check_unresolved: cannot find dependency samba4-libs for samba4-server
 * pkg_hash_fetch_best_installation_candidate: Packages for samba4-server found, but incompatible with the architectures configured
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for autosamba:
 * 	samba4-libs
 * opkg_install_cmd: Cannot install package autosamba.
 * check_conflicts_for: The following packages conflict with kmod-fast-classifier:
 * check_conflicts_for: 	kmod-shortcut-fe-cm * 
 * opkg_install_cmd: Cannot install package luci-app-turboacc.
 * check_conflicts_for: The following packages conflict with samba4-server:
 * check_conflicts_for: 	samba36-server * 
 * opkg_install_cmd: Cannot install package samba4-server.
make[2]: *** [package/Makefile:70: package/install] Error 255
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:111: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
Error: Process completed with exit code 2.
---

wggglggg, 2022-10-02:
去Luci-application   里的luci-app-samba钩去掉，保留samba4
再去network 里面samaba36-server钩去掉。
然后rm -rf ./tmp与 ./scripts/feeds update -a 和  ./scripts/feeds install -a和make download -j8（这一行做不做似乎没必要，我是做了）
再用单线程编译make V=s -j1，这样就避开samba36与samba4冲突的情况。
我自己试过编译是成功的，还没重写到软路由中，应该没什么问题。

closed 2022-10-04 by kenzok8