kenzok8/openwrt-packages #342: 编译X86的目标文件Legancy到达ddnsto那出错

liuhejun2020, 2022-11-07:
编译X86的目标文件选传统模式Legancy时到达ddnsto那出错，选其他固件目标文件时能过

shenyb47, 2022-11-07:
日志如下：
rm -rf /workdir/openwrt/build_dir/target-i386_pentium-mmx_musl/ddnsto-binary-3.0.2/.pkgdir/ddnsto.installed /workdir/openwrt/build_dir/target-i386_pentium-mmx_musl/ddnsto-binary-3.0.2/.pkgdir/ddnsto
mkdir -p /workdir/openwrt/build_dir/target-i386_pentium-mmx_musl/ddnsto-binary-3.0.2/.pkgdir/ddnsto
install -d -m0755 /workdir/openwrt/build_dir/target-i386_pentium-mmx_musl/ddnsto-binary-3.0.2/.pkgdir/ddnsto/usr/sbin /workdir/openwrt/build_dir/target-i386_pentium-mmx_musl/ddnsto-binary-3.0.2/.pkgdir/ddnsto/etc/config /workdir/openwrt/build_dir/target-i386_pentium-mmx_musl/ddnsto-binary-3.0.2/.pkgdir/ddnsto/etc/init.d /workdir/openwrt/build_dir/target-i386_pentium-mmx_musl/ddnsto-binary-3.0.2/.pkgdir/ddnsto/etc/uci-defaults
install -m0755 /workdir/openwrt/build_dir/target-i386_pentium-mmx_musl/ddnsto-binary-3.0.2/ddnsto.i386 /workdir/openwrt/build_dir/target-i386_pentium-mmx_musl/ddnsto-binary-3.0.2/.pkgdir/ddnsto/usr/sbin/ddnstod
install: cannot stat '/workdir/openwrt/build_dir/target-i386_pentium-mmx_musl/ddnsto-binary-3.0.2/ddnsto.i386': No such file or directory
make[3]: *** [Makefile:70: /workdir/openwrt/build_dir/target-i386_pentium-mmx_musl/ddnsto-binary-3.0.2/.pkgdir/ddnsto.installed] Error 1
make[3]: Leaving directory '/workdir/openwrt/feeds/kenzo/ddnsto'
time: package/feeds/kenzo/ddnsto/compile#0.20#0.06#0.24
    ERROR: package/feeds/kenzo/ddnsto failed to build.
make[2]: *** [package/Makefile:116: package/feeds/kenzo/ddnsto/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-i386_pentium-mmx_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
Error: Process completed with exit code 2.

closed 2023-01-30 by kenzok8

arbaleast, 2023-04-12:
```
tar -tvf dl/ddnsto-binary-3.0.2.tar.gz 
drwxrwxr-x janson/janson     0 2022-08-10 19:53 ddnsto-binary-3.0.2/
-rwxrwxr-x janson/janson 1824208 2022-08-10 19:53 ddnsto-binary-3.0.2/ddwebdav.mipsel
-rwxrwxr-x janson/janson 2196424 2022-08-10 19:53 ddnsto-binary-3.0.2/ddwebdav.x86_64
-rwxrwxr-x janson/janson  843608 2022-08-10 19:53 ddnsto-binary-3.0.2/ddnsto.x86_64
-rwxrwxr-x janson/janson  564012 2022-08-10 19:53 ddnsto-binary-3.0.2/ddnsto.arm
-rwxrwxr-x janson/janson 1688344 2022-08-10 19:53 ddnsto-binary-3.0.2/ddwebdav.aarch64
-rwxr-xr-x janson/janson  554188 2022-08-10 19:53 ddnsto-binary-3.0.2/ddnsto.mipsel
-rwxrwxr-x janson/janson 1626528 2022-08-10 19:53 ddnsto-binary-3.0.2/ddnsto.aarch64
-rwxrwxr-x janson/janson 1214212 2022-08-10 19:53 ddnsto-binary-3.0.2/ddwebdav.arm
```
ddnsto 是不支持i386架构吗？