kenzok8/openwrt-packages #356: mosdns缺少依赖项upx/host

handsomexdd1024, 2023-01-10:
使用openwrt官方源openwrt-22.03分支源码执行`make menuconfig`时，提示：
`WARNING: Makefile 'package/feeds/kenzok/mosdns/Makefile' has a build dependency on 'upx/host', which does not exist`
请检查是否缺少[upx/upx](https://github.com/kuoruan/openwrt-upx)依赖项？

kenzok8, 2023-01-24:
sudo apt-get -y install build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev patch python3 python2.7 unzip zlib1g-dev lib32gcc-s1 libc6-dev-i386 subversion flex uglifyjs gcc-multilib g++-multilib p7zip p7zip-full msmtp libssl-dev texinfo libglib2.0-dev xmlto qemu-utils libelf-dev autoconf automake libtool autopoint device-tree-compiler ccache xsltproc rename antlr3 gperf curl screen upx-ucl jq

handsomexdd1024, 2023-01-26:
哦这样，所以你的意思是在编译的主机上安装ucl嘛
行我再试试，非常感谢

closed 2023-01-26 by kenzok8

ramaox, 2023-02-18:
git clone https://github.com/kuoruan/openwrt-upx.git package/openwrt-upx

ramaox, 2023-02-18:
`git clone https://github.com/kuoruan/openwrt-upx.git package/openwrt-upx`

ramaox, 2023-02-18:
`cd openwrt`
`git clone https://github.com/kuoruan/openwrt-upx.git package/openwrt-upx`

ramaox, 2023-02-18:
`cd openwrt
git clone https://github.com/kuoruan/openwrt-upx.git package/openwrt-upx`

kenzok8, 2023-02-18:
upx内核我删除了，因为现在编译不会报错了