kenzok8/openwrt-packages #362: error in first code

mmpma, 2023-01-29:
why i get this error?
root@X-WRT:~# sed -i '$a src-git small https://github.com/kenzok8/small' feeds.conf.default
sed: feeds.conf.default: No such file or director

there isn't any toturial for install on xwrt?

HollyMadCow, 2023-01-30:
edit file feeds.conf.default
add   

src-git small https://github.com/kenzok8/small

to the last line 

then run ./scripts/feeds update -a && ./scripts/feeds install -a

mmpma, 2023-01-30:
> edit file feeds.conf.default add
> 
> src-git small https://github.com/kenzok8/small
> 
> to the last line
> 
> then run ./scripts/feeds update -a && ./scripts/feeds install -a

where is the location of  file feeds.conf.default

kenzok8, 2023-01-30:
is this repo？
https://github.com/x-wrt/x-wrt
[image]

kenzok8, 2023-01-31:
Compile is not a root user, “feeds.conf.default” is in the wrt directory

closed 2023-02-01 by kenzok8