kenzok8/openwrt-packages #363: alist编译错误

vinwinchen, 2023-01-30:
github.com/alist-org/alist/v3/pkg/gowebdav/cmd/gowebdav
github.com/alist-org/alist/v3/internal/search
github.com/aws/aws-sdk-go/service/s3/s3manager
github.com/alist-org/alist/v3/server/handles
github.com/alist-org/alist/v3/drivers/123
github.com/alist-org/alist/v3/drivers/mediatrack
github.com/alist-org/alist/v3/drivers/pikpak
github.com/alist-org/alist/v3/drivers/s3
github.com/alist-org/alist/v3/drivers/thunder
github.com/alist-org/alist/v3/server
github.com/alist-org/alist/v3/drivers
github.com/jackc/pgx/v5
github.com/jackc/pgx/v5/stdlib
gorm.io/driver/postgres
gorm.io/driver/sqlite
github.com/alist-org/alist/v3/internal/bootstrap
github.com/alist-org/alist/v3/cmd
github.com/alist-org/alist/v3

make[3]: *** [Makefile:102: /home/make/lede/build_dir/target-x86_64_musl/alist-3.9.2/.built] Error 2
make[3]: Leaving directory '/home/make/lede/feeds/kenzo/alist'
time: package/feeds/kenzo/alist/compile#81.65#15.58#57.57
    ERROR: package/feeds/kenzo/alist failed to build.
make[2]: *** [package/Makefile:116: package/feeds/kenzo/alist/compile] Error 1
make[2]: Leaving directory '/home/make/lede'
make[1]: *** [package/Makefile:110: /home/make/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/make/lede'
make: *** [/home/make/lede/include/toplevel.mk:231: world] Error 2

jinjin327, 2023-01-31:
遇到一样的问题

ThingsWhy, 2023-02-01:
我这里也是，我单独编译luci-app-alist的时候第一行出来了关于mosdns的依赖信息[image]

vinwinchen, 2023-02-02:
``Building targets
github.com/winfsp/cgofuse/fuse
# github.com/winfsp/cgofuse/fuse
../../../../../dl/go-mod-cache/github.com/winfsp/cgofuse@v1.5.0/fuse/host_cgo.go:116:10: fatal error: fuse.h: No such file or directory
  116 | #include <fuse.h>
      |          ^~~~~~~~
compilation terminated.

make[3]: *** [Makefile:102: /home/make/lede/build_dir/target-x86_64_musl/alist-3.9.2/.built] Error 2
make[3]: Leaving directory '/home/make/lede/feeds/kenzo/alist'
time: package/feeds/kenzo/alist/compile#1.83#0.63#1.33
    ERROR: package/feeds/kenzo/alist failed to build.
make[2]: *** [package/Makefile:116: package/feeds/kenzo/alist/compile] Error 1
make[2]: Leaving directory '/home/make/lede'
make[1]: *** [package/Makefile:110: /home/make/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/make/lede'
make: *** [/home/make/lede/include/toplevel.mk:231: world] Error 2

xinaz, 2023-02-06:
碰到一样的问题
Building targets
github.com/winfsp/cgofuse/fuse
# github.com/winfsp/cgofuse/fuse
../../../../../dl/go-mod-cache/github.com/winfsp/cgofuse@v1.5.0/fuse/host_cgo.go:116:10: fatal error: fuse.h: No such file or directory
  116 | #include <fuse.h>
      |          ^~~~~~~~
compilation terminated.

make[3]: *** [Makefile:102: /home/xinaz/project/lede/build_dir/target-arm_cortex-a9_musl_eabi/alist-3.10.1/.built] Error 2
make[3]: Leaving directory '/home/xinaz/project/lede/feeds/kenzok/alist'
time: package/feeds/kenzok/alist/compile#4.99#1.76#2.85
    ERROR: package/feeds/kenzok/alist failed to build.
make[2]: *** [package/Makefile:116: package/feeds/kenzok/alist/compile] Error 1
make[2]: Leaving directory '/home/xinaz/project/lede'
make[1]: *** [package/Makefile:110: /home/xinaz/project/lede/staging_dir/target-arm_cortex-a9_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/xinaz/project/lede'
make: *** [/home/xinaz/project/lede/include/toplevel.mk:231: world] Error 2
[xinaz@localhost lede]$ make menuconfig
configuration written to .config

dudulongg, 2023-02-07:
我也是同样的问题，等待作者修正吧

kenzok8, 2023-02-07:
参考作者的提示
更新 golang 19.x

rm -rf feeds/packages/lang/golang
svn export https://github.com/sbwml/packages_lang_golang/branches/19.x feeds/packages/lang/golang

ThingsWhy, 2023-02-08:
确实还是云编译失败了

fseteen, 2023-02-09:
> 参考作者的提示 更新 golang 19.x
> 
> rm -rf feeds/packages/lang/golang svn export https://github.com/sbwml/packages_lang_golang/branches/19.x feeds/packages/lang/golang

编译成功！

dudulongg, 2023-02-09:
我重装Ubuntu从新配置menuconfig就好了

kenzok8, 2023-02-10:
> 删掉lede目录都不行，重装Ubuntu从新配置menuconfig就好了

精神可嘉，折腾不止

kvmcore, 2023-02-11:
> 参考作者的提示 更新 golang 19.x
> 
> rm -rf feeds/packages/lang/golang svn export https://github.com/sbwml/packages_lang_golang/branches/19.x feeds/packages/lang/golang

装上fuse开发包，问题已解决。

ubuntu/debian:
sudo apt install libfuse-dev

CentOS:
sudo yum install fuse-devel

vinwinchen, 2023-02-11:
> > 参考作者的提示 更新 golang 19.x
> > rm -rf feeds/packages/lang/golang svn export https://github.com/sbwml/packages_lang_golang/branches/19.x feeds/packages/lang/golang
> 
> 装上fuse开发包，问题已解决。
> 
> ubuntu/debian: sudo apt install libfuse-dev
> 
> CentOS: sudo yum install fuse-devel

这个方法最好用

closed 2023-02-11 by vinwinchen

michealhansun, 2023-05-23:
> > 参考作者的提示 更新 golang 19.x
> > rm -rf feeds/packages/lang/golang svn export https://github.com/sbwml/packages_lang_golang/branches/19.x feeds/packages/lang/golang
> 
> 装上fuse开发包，问题已解决。
> 
> ubuntu/debian: sudo apt install libfuse-dev
> 
> CentOS: sudo yum install fuse-devel

你好，请问github actions 云编译时如何添加命令安装

ThingsWhy, 2023-05-24:
> > > 参考作者的提示 更新 golang 19.x
> > > rm -rf feeds/packages/lang/golang svn export https://github.com/sbwml/packages_lang_golang/branches/19.x feeds/packages/lang/golang
> > 
> > 
> > 装上fuse开发包，问题已解决。
> > ubuntu/debian: sudo apt install libfuse-dev
> > CentOS: sudo yum install fuse-devel
> 
> 你好，请问github actions 云编译时如何添加命令安装

类似这样
[image]
不过我现在似乎不需要了，用的那个分支已经更新了依赖了