kenzok8/openwrt-packages #367: xray-core failed to build

xinaz, 2023-02-01:
make[3]: *** [Makefile:90: /home/xinaz/project/lede_make/build_dir/target-arm_cortex-a9_musl_eabi/Xray-core-1.7.2/.built] Error 1
make[3]: Leaving directory '/home/xinaz/project/lede_make/feeds/small/xray-core'
time: package/feeds/small/xray-core/compile#2.32#2.36#6.59
    ERROR: package/feeds/small/xray-core failed to build.
make[2]: *** [package/Makefile:116: package/feeds/small/xray-core/compile] Error 1
make[2]: Leaving directory '/home/xinaz/project/lede_make'
make[1]: *** [package/Makefile:110: /home/xinaz/project/lede_make/staging_dir/target-arm_cortex-a9_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/xinaz/project/lede_make'
make: *** [/home/xinaz/project/lede_make/include/toplevel.mk:231: world] Error 2

dukmonki, 2023-02-01:
一样的问题。

kenzok8, 2023-02-01:
git pull && ./scripts/feeds update -a && ./scripts/feeds install -a

重新拉取下，应该是补丁文件的问题

xinaz, 2023-02-02:
昨晚又重新试过几次，还不行。

make[3]: *** [Makefile:52: /home/xinaz/project/lede_make/build_dir/target-arm_cortex-a9_musl_eabi/Xray-core-1.7.2/.built] Error 1
make[3]: Leaving directory '/home/xinaz/project/lede_make/feeds/small/xray-core'
time: package/feeds/small/xray-core/compile#1.44#0.58#2.44
    ERROR: package/feeds/small/xray-core failed to build.
make[2]: *** [package/Makefile:116: package/feeds/small/xray-core/compile] Error 1
make[2]: Leaving directory '/home/xinaz/project/lede_make'
make[1]: *** [package/Makefile:110: /home/xinaz/project/lede_make/staging_dir/target-arm_cortex-a9_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/xinaz/project/lede_make'
make: *** [/home/xinaz/project/lede_make/include/toplevel.mk:231: world] Error 2

xinaz, 2023-02-02:
../../infra/conf/serial/loader.go:8:2: github.com/ghodss/yaml@v1.0.1-0.20190212211648-25d852aebe32: Get "https://proxy.golang.org/github.com/ghodss/yaml/@v/v1.0.1-0.20190212211648-25d852aebe32.zip": proxyconnect tcp: EOF
../../common/serial/typed_message.go:7:2: github.com/golang/protobuf@v1.5.2: Get "https://proxy.golang.org/github.com/golang/protobuf/@v/v1.5.2.zip": proxyconnect tcp: EOF
../../transport/internet/websocket/connection.go:8:2: github.com/gorilla/websocket@v1.5.0: Get "https://proxy.golang.org/github.com/gorilla/websocket/@v/v1.5.0.zip": proxyconnect tcp: EOF

xinaz, 2023-02-02:
重拉之后，问题一样，不过在前面的log里看到一些以下问题，用的是代理访问国外网站。
../../infra/conf/serial/loader.go:8:2: github.com/ghodss/yaml@v1.0.1-0.20190212211648-25d852aebe32: Get "https://proxy.golang.org/github.com/ghodss/yaml/@v/v1.0.1-0.20190212211648-25d852aebe32.zip": proxyconnect tcp: EOF
../../common/serial/typed_message.go:7:2: github.com/golang/protobuf@v1.5.2: Get "https://proxy.golang.org/github.com/golang/protobuf/@v/v1.5.2.zip": proxyconnect tcp: EOF
../../transport/internet/websocket/connection.go:8:2: github.com/gorilla/websocket@v1.5.0: Get "https://proxy.golang.org/github.com/gorilla/websocket/@v/v1.5.0.zip": proxyconnect tcp: EOF

xinaz, 2023-02-02:
代理问题，问题结束！多谢！

closed 2023-02-02 by kenzok8