kenzok8/openwrt-packages #374: luci-app-adguardhome编译时没有选择编译内核的选项

gooico, 2023-02-23:
升级后没有这个选项了，无法编译时添加内核，这个功能很需要。直接编译增加内核固件占用空间小，后续再更新内核容易失败。

ThingsWhy, 2023-02-23:
在network里还有这个东西，你可以选了试试
[image]

ThingsWhy, 2023-02-23:
不过还有个疑问，现在不编译adgh在luci里默认的保存路径是tmp，编译了就不知道会怎么样了，不知道是不是还是放到tmp

gooico, 2023-02-27:
> 在网络里还有这个东西，之前luci里选了好看这里就会选中的，你可以选了试试 [image]

感谢回复，这个路径选中之后固件大小增加了9M，但实机刷入后仍显示无内核。

ThingsWhy, 2023-02-27:
你好，我刚刚编译了一遍看了下，他似乎把文件放到了/usr/bin/AdGuardHome，但是他的配置文件adguardhome.init我猜可能没有生效，还有一点，luci的工作目录默认就是/usr/bin/AdGuardHome，我这报了错，改了就能识别到bin文件了

kenzok8, 2023-03-01:
https://github.com/kenzok8/AutoSync/commit/4e77ac2dc74bcc59df8fd4ce5e21d689da949b75

ThingsWhy, 2023-03-02:
> [kenzok8/AutoSync@4e77ac2](https://github.com/kenzok8/AutoSync/commit/4e77ac2dc74bcc59df8fd4ce5e21d689da949b75)

大佬，还得麻烦你一下，我刚刚编译了一遍，发现不知道为什么一开始就创建了一个/usr/bin/AdGuardHome的目录导致bin文件没法放到这里，他自动放到下一级了，还需要修改执行文件路径才能应用
[image]

kenzok8, 2023-03-03:
没看明白什么意思

Trucluster, 2023-03-28:
目前解决方案，执行
uci set AdGuardHome.AdGuardHome.binpath="/usr/bin/AdGuardHome/AdGuardHome"
uci set AdGuardHome.AdGuardHome.update_url="https://github.com/AdguardTeam/AdGuardHome/releases/download/v0.107.26/AdGuardHome_linux_amd64.tar.gz"
reboot
进入AdGuard Home界面，点击更新核心，将自动下载核心，不需要重新编译了