kenzok8/openwrt-packages #378: 最新naiveproxy编译失败，提示缺少gn/host

mahakalaya, 2023-03-18:
make -j1 V=s package/naiveproxy/compile
WARNING: Makefile 'package/feeds/kenzok/naiveproxy/Makefile' has a build dependency on 'gn/host', which does not exist
make[2]: Entering directory '/home/simon/lede/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/home/simon/lede/scripts/config'
bash: line 1: *1024*1024: syntax error: operand expected (error token is "*1024*1024")
time: target/linux/prereq#0.12#0.15#0.25
Checking 'rsync'... ok.
make[1]: Entering directory '/home/simon/lede'
make[2]: Entering directory '/home/simon/lede/package/libs/toolchain'
echo "libc" >> /home/simon/lede/staging_dir/target-aarch64_cortex-a53_musl/pkginfo/toolchain.default.install
echo "libgcc" >> /home/simon/lede/staging_dir/target-aarch64_cortex-a53_musl/pkginfo/toolchain.default.install
echo "libatomic" >> /home/simon/lede/staging_dir/target-aarch64_cortex-a53_musl/pkginfo/toolchain.default.install
echo "libstdcpp" >> /home/simon/lede/staging_dir/target-aarch64_cortex-a53_musl/pkginfo/toolchain.default.install
echo "libpthread" >> /home/simon/lede/staging_dir/target-aarch64_cortex-a53_musl/pkginfo/toolchain.default.install
echo "librt" >> /home/simon/lede/staging_dir/target-aarch64_cortex-a53_musl/pkginfo/toolchain.default.install
make[2]: Leaving directory '/home/simon/lede/package/libs/toolchain'
time: package/libs/toolchain/compile#0.16#0.28#0.39
make[2]: Entering directory '/home/simon/lede/feeds/kenzok/naiveproxy'
rm -f /home/simon/lede/build_dir/target-aarch64_cortex-a53_musl/naiveproxy-111.0.5563.64-1/.built
touch /home/simon/lede/build_dir/target-aarch64_cortex-a53_musl/naiveproxy-111.0.5563.64-1/.built_check
ninja -C "/home/simon/lede/build_dir/target-aarch64_cortex-a53_musl/naiveproxy-111.0.5563.64-1/src/out/Release" naive
ninja: Entering directory `/home/simon/lede/build_dir/target-aarch64_cortex-a53_musl/naiveproxy-111.0.5563.64-1/src/out/Release'
ninja: fatal: chdir to '/home/simon/lede/build_dir/target-aarch64_cortex-a53_musl/naiveproxy-111.0.5563.64-1/src/out/Release' - No such file or directory
make[2]: *** [Makefile:119: /home/simon/lede/build_dir/target-aarch64_cortex-a53_musl/naiveproxy-111.0.5563.64-1/.built] Error1
make[2]: Leaving directory '/home/simon/lede/feeds/kenzok/naiveproxy'
time: package/feeds/kenzok/naiveproxy/compile#0.29#0.27#0.50
    ERROR: package/feeds/kenzok/naiveproxy failed to build.
make[1]: *** [package/Makefile:116: package/feeds/kenzok/naiveproxy/compile] Error 1
make[1]: Leaving directory '/home/simon/lede'
make: *** [/home/simon/lede/include/toplevel.mk:231：package/naiveproxy/compile] 错误 2

xuexijiaoben, 2023-03-18:
同样报错，应该是上游没有同步完整，上次alist也是，就是不知道这个上游在哪儿

xyl94k, 2023-03-18:
同样的问题+1

mahakalaya, 2023-03-19:
刚刚通过了 改回去了 成老样子了

kenzok8, 2023-03-19:
> 刚刚通过了 改回去了 成老样子了

git pull 一下吧

gnusir, 2023-03-19:
> > 刚刚通过了 改回去了 成老样子了
> 
> git pull 一下吧

还是那个提示

kenzok8, 2023-03-19:
> > > 刚刚通过了 改回去了 成老样子了
> > 
> > 
> > git pull 一下吧
> 
> 还是那个提示
./scripts/feeds clean
./scripts/feeds update -a
./scripts/feeds install -a

la8896, 2023-03-19:
Collecting package info: package/feeds/packages/udp-broadcast-relay-redux-openwrCollecting package info: done
WARNING: Makefile 'package/feeds/kenzo/naiveproxy/Makefile' has a build dependency on 'gn/host', which does not exist

mahakalaya, 2023-03-19:
上上次更新那个可以

mahakalaya, 2023-03-19:
> Reference in new i

上午9e738e47833a68df8401bba32528efa42d7c0172这个提交编译正常
编译arm64固件
编译电脑是debian11

kenzok8, 2023-03-19:
commit/9e738e47833a68df8401bba32528efa42d7c0172
是旧版
新releases #https://github.com/klzgrad/naiveproxy/releases
上游是#https://github.com/fw876/helloworld/tree/master/naiveproxy

closed 2023-03-20 by kenzok8