kenzok8/openwrt-packages #379: 编译alist出问题

JackyLeeGM, 2023-03-18:
1.搜索了该问题，没有类似的issue
2.详细叙述：（1）具体问题：每次alist更新，在编译就会出现问题，编译失败；（2）路由器型号和固件版本：NanoPi R5S，固件版本Lean内核5.15
3.详细日志：
make[3]: *** [Makefile:72: /home/jackylee/lede/build_dir/target-aarch64_generic_musl/alist-3.14.0/.built] Error 2
make[3]: Leaving directory '/home/jackylee/lede/feeds/kenzo/alist'
time: package/feeds/kenzo/alist/compile#96.18#24.70#64.47
    ERROR: package/feeds/kenzo/alist failed to build.
make[2]: *** [package/Makefile:116: package/feeds/kenzo/alist/compile] Error 1
make[2]: Leaving directory '/home/jackylee/lede'
make[1]: *** [package/Makefile:110: /home/jackylee/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/jackylee/lede'
make: *** [/home/jackylee/lede/include/toplevel.mk:231：world] 错误 2

求教 大佬指点迷津

kenzok8, 2023-03-18:
https://github.com/sbwml/luci-app-alist
成功编译过吗？

JackyLeeGM, 2023-03-18:
> https://github.com/sbwml/luci-app-alist 成功编译过吗？

成功过

JackyLeeGM, 2023-03-18:
> https://github.com/sbwml/luci-app-alist 成功编译过吗？

就像内核从5.15.98到5.15.102这种，加上alist插件作者刚更新，就会出现这种情况，删掉拉取的所有alist，在编辑就成功

kenzok8, 2023-03-18:
> 就会出现这种情况，删掉拉取的所有alist，在编辑就成功

我的意思是以前有成功编译过alist吗？

JackyLeeGM, 2023-03-19:
> > 就会出现这种情况，删掉拉取的所有alist，在编辑就成功
> 
> 我的意思是以前有成功编译过alist吗？

也成功过，所以就是不晓得哪里出了问题，需要大佬指点一下了

kenzok8, 2023-03-19:
> > > 就会出现这种情况，删掉拉取的所有alist，在编辑就成功
> > 
> > 
> > 我的意思是以前有成功编译过alist吗？
> 
> 也成功过，所以就是不晓得哪里出了问题，需要大佬指点一下了我
上面我给了作者的仓库地址，他那有说明，因为你也没提供更详细的信息，我估计是golang的问题

JackyLeeGM, 2023-03-19:
> > > > 就会出现这种情况，删掉拉取的所有alist，在编辑就成功
> > > 
> > > 
> > > 我的意思是以前有成功编译过alist吗？
> > 
> > 
> > 也成功过，所以就是不晓得哪里出了问题，需要大佬指点一下了我
> > 上面我给了作者的仓库地址，他那有说明，因为你也没提供更详细的信息，我估计是golang的问题

感谢kenzok8 大佬回复，我再研究下，谢谢了

JackyLeeGM, 2023-03-19:
> > > > 就会出现这种情况，删掉拉取的所有alist，在编辑就成功
> > > 
> > > 
> > > 我的意思是以前有成功编译过alist吗？
> > 
> > 
> > 也成功过，所以就是不晓得哪里出了问题，需要大佬指点一下了我
> > 上面我给了作者的仓库地址，他那有说明，因为你也没提供更详细的信息，我估计是golang的问题

make[8]: Leaving directory '/home/jackylee/lede/build_dir/hostpkg/gettext-0.21.1/gettext-tools/projects'
make[7]: Leaving directory '/home/jackylee/lede/build_dir/hostpkg/gettext-0.21.1/gettext-tools/projects'
Making install in styles
make[7]: Entering directory '/home/jackylee/lede/build_dir/hostpkg/gettext-0.21.1/gettext-tools/styles'
make[8]: Entering directory '/home/jackylee/lede/build_dir/hostpkg/gettext-0.21.1/gettext-tools/styles'
make[8]: Nothing to be done for 'install-exec-am'.
 /usr/bin/mkdir -p '/home/jackylee/lede/staging_dir/hostpkg/share/gettext/styles'
 /home/jackylee/lede/staging_dir/host/bin/install -c -m 644 po-default.css po-vim.css po-emacs-x.css po-emacs-xterm256.css po-emacs-xterm16.css po-emacs-xterm.css '/home/jackylee/lede/staging_dir/hostpkg/share/gettext/styles'
make[8]: Leaving directory '/home/jackylee/lede/build_dir/hostpkg/gettext-0.21.1/gettext-tools/styles'
make[7]: Leaving directory '/home/jackylee/lede/build_dir/hostpkg/gettext-0.21.1/gettext-tools/styles'
Making install in misc
make[7]: Entering directory '/home/jackylee/lede/build_dir/hostpkg/gettext-0.21.1/gettext-tools/misc'
make[8]: Entering directory '/home/jackylee/lede/build_dir/hostpkg/gettext-0.21.1/gettext-tools/misc'
 /usr/bin/mkdir -p '/home/jackylee/lede/staging_dir/hostpkg/bin'
 /home/jackylee/lede/staging_dir/host/bin/install -c gettextize autopoint '/home/jackylee/lede/staging_dir/hostpkg/bin'
 /usr/bin/mkdir -p '/home/jackylee/lede/staging_dir/hostpkg/share/gettext'
 /home/jackylee/lede/staging_dir/host/bin/install -c -m 644 archive.dir.tar.xz '/home/jackylee/lede/staging_dir/hostpkg/share/gettext'
make[8]: Leaving directory '/home/jackylee/lede/build_dir/hostpkg/gettext-0.21.1/gettext-tools/misc'
make[7]: Leaving directory '/home/jackylee/lede/build_dir/hostpkg/gettext-0.21.1/gettext-tools/misc'
Making install in m4
make[7]: Entering directory '/home/jackylee/lede/build_dir/hostpkg/gettext-0.21.1/gettext-tools/m4'
make[8]: Entering directory '/home/jackylee/lede/build_dir/hostpkg/gettext-0.21.1/gettext-tools/m4'
make[8]: Nothing to be done for 'install-exec-am'.
 /usr/bin/mkdir -p '/home/jackylee/lede/staging_dir/hostpkg/share/aclocal'
 /home/jackylee/lede/staging_dir/host/bin/install -c -m 644 ../../gettext-runtime/gnulib-m4/host-cpu-c-abi.m4 ../../gettext-runtime/gnulib-m4/lib-ld.m4 ../../gettext-runtime/gnulib-m4/lib-link.m4 ../../gettext-runtime/gnulib-m4/lib-prefix.m4 ../../gettext-runtime/m4/gettext.m4 ../../gettext-runtime/m4/iconv.m4 ../../gettext-runtime/m4/intlmacosx.m4 ../../gettext-runtime/m4/nls.m4 ../../gettext-runtime/m4/po.m4 ../../gettext-runtime/m4/progtest.m4 '/home/jackylee/lede/staging_dir/hostpkg/share/aclocal'
 /usr/bin/mkdir -p '/home/jackylee/lede/staging_dir/hostpkg/share/doc/gettext/examples/build-aux'
 /home/jackylee/lede/staging_dir/host/bin/install -c -m 644 ../gnulib-m4/javacomp.m4 ../gnulib-m4/javaexec.m4 ../gnulib-m4/csharpcomp.m4 ../gnulib-m4/csharpexec.m4 csharpexec-test.exe '/home/jackylee/lede/staging_dir/hostpkg/share/doc/gettext/examples/build-aux'
make[8]: Leaving directory '/home/jackylee/lede/build_dir/hostpkg/gettext-0.21.1/gettext-tools/m4'
make[7]: Leaving directory '/home/jackylee/lede/build_dir/hostpkg/gettext-0.21.1/gettext-tools/m4'
make[7]: Entering directory '/home/jackylee/lede/build_dir/hostpkg/gettext-0.21.1/gettext-tools'
make[8]: Entering directory '/home/jackylee/lede/build_dir/hostpkg/gettext-0.21.1/gettext-tools'
make[8]: Nothing to be done for 'install-exec-am'.
 /usr/bin/mkdir -p '/home/jackylee/lede/staging_dir/hostpkg/share/doc/gettext/examples/build-aux'
 /home/jackylee/lede/staging_dir/host/bin/install -c -m 644 ../build-aux/javacomp.sh.in ../build-aux/javaexec.sh.in ../build-aux/csharpcomp.sh.in ../build-aux/csharpexec.sh.in '/home/jackylee/lede/staging_dir/hostpkg/share/doc/gettext/examples/build-aux'
 /usr/bin/mkdir -p '/home/jackylee/lede/staging_dir/hostpkg/share/gettext'
 /home/jackylee/lede/staging_dir/host/bin/install -c ../build-aux/config.rpath '/home/jackylee/lede/staging_dir/hostpkg/share/gettext'
make[8]: Leaving directory '/home/jackylee/lede/build_dir/hostpkg/gettext-0.21.1/gettext-tools'
make[7]: Leaving directory '/home/jackylee/lede/build_dir/hostpkg/gettext-0.21.1/gettext-tools'
make[6]: Leaving directory '/home/jackylee/lede/build_dir/hostpkg/gettext-0.21.1/gettext-tools'
make[6]: Entering directory '/home/jackylee/lede/build_dir/hostpkg/gettext-0.21.1'
make[7]: Entering directory '/home/jackylee/lede/build_dir/hostpkg/gettext-0.21.1'
make[7]: Nothing to be done for 'install-exec-am'.
make[7]: Nothing to be done for 'install-data-am'.
make[7]: Leaving directory '/home/jackylee/lede/build_dir/hostpkg/gettext-0.21.1'
make[6]: Leaving directory '/home/jackylee/lede/build_dir/hostpkg/gettext-0.21.1'
make[5]: Leaving directory '/home/jackylee/lede/build_dir/hostpkg/gettext-0.21.1'
make[4]: Leaving directory '/home/jackylee/lede/build_dir/hostpkg/gettext-0.21.1'
ln -sf msgfmt /home/jackylee/lede/staging_dir/hostpkg/bin/gmsgfmt
mkdir -p /home/jackylee/lede/staging_dir/hostpkg/stamp
touch /home/jackylee/lede/build_dir/hostpkg/gettext-0.21.1/.built
touch /home/jackylee/lede/staging_dir/hostpkg/stamp/.gettext-full_installed
make[3]: Leaving directory '/home/jackylee/lede/package/libs/gettext-full'
time: package/libs/gettext-full/host-compile#145.46#27.10#288.87
make[2]: Leaving directory '/home/jackylee/lede'
make[1]: *** [package/Makefile:110: /home/jackylee/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/jackylee/lede'
make: *** [/home/jackylee/lede/include/toplevel.mk:231：world] 错误 2

JackyLeeGM, 2023-03-19:
> make[8]: Leaving directory '/home/jackylee/lede/build_dir/hostpkg/gettext-0.21.1/gettext-tools/projects' make[7]: Leaving directory [...]

这是新一轮错误，不知道是否能更详细的了解，麻烦大佬再看看呢

d975xbx, 2023-03-19:
遇到一样的错误。alist

dudulongg, 2023-03-19:
正常编译3.14.0-1版本，没有错误

JackyLeeGM, 2023-03-20:
> 

是意思说我拉取的alist的版本不对？

kenzok8, 2023-03-20:
不是版本问题，是golang版本，今天L大的已经更新了，如果是1.18版本alist编译过不了

closed 2023-03-22 by kenzok8

xpfoxs, 2023-03-25:
安装 libfuse-dev 就好了