kenzok8/openwrt-packages #384: naiveproxy编译问题

hxx006, 2023-04-01:
time: package/feeds/kenzo/naiveproxy/compile#0.43#0.17#0.71
    ERROR: package/feeds/kenzo/naiveproxy failed to build.
make[2]: *** [package/Makefile:116: package/feeds/kenzo/naiveproxy/compile] Error 1
make[2]: *** Waiting for unfinished jobs....
mkdir -p /home/hxx006/lede/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/hxx006/lede/tmp/.root-copy.flock -c 'cp -fpR /home/hxx006/lede/build_dir/target-x86_64_musl/libev-4.33/.pkgdir/libev/. /home/hxx006/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/hxx006/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.libev_installed
echo "libev" >> /home/hxx006/lede/staging_dir/target-x86_64_musl/pkginfo/libev.default.install
make[3]: Leaving directory '/home/hxx006/lede/feeds/packages/libs/libev'
time: package/feeds/packages/libev/compile#0.23#0.13#0.43
make[2]: Leaving directory '/home/hxx006/lede'
make[1]: *** [package/Makefile:110: /home/hxx006/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/hxx006/lede'
make: *** [/home/hxx006/lede/include/toplevel.mk:231：world] 错误 2

OnlyCharacters, 2023-04-01:
我也遇到了
```
make[3]: Entering directory '/home/debian/openwrt/feeds/kenzo/naiveproxy'
touch /home/debian/openwrt/build_dir/target-aarch64_cortex-a72_musl/naiveproxy-111.0.5563.64-1/.prepared_00a9029f8b59ffea7ad6256ad07384c5_6664517399ebbbc92a37c5bb081b5c53_check
. /home/debian/openwrt/include/shell.sh; gzip -dc /home/debian/openwrt/dl/naiveproxy-111.0.5563.64-1.tar.gz | tar -C /home/debian/openwrt/build_dir/target-aarch64_cortex-a72_musl/naiveproxy-111.0.5563.64-1/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/. /home/debian/openwrt/build_dir/target-aarch64_cortex-a72_musl/naiveproxy-111.0.5563.64-1

Applying ./patches/100-macros.patch using plaintext: 
patching file src/base/allocator/partition_allocator/tagging.cc
( pushd /home/debian/openwrt/build_dir/target-aarch64_cortex-a72_musl/naiveproxy-111.0.5563.64-1/src ; mkdir -p "chrome/build/pgo_profiles" ; cp -fpR "/home/debian/openwrt/dl/chrome-linux-5563-1677886924-aa8d450a18b4f7f1cd41619f60bc5da6beb3c1d0.profdata" "chrome/build/pgo_profiles" ; mkdir -p "third_party/llvm-build/Release+Asserts" ; tar -xzf "/home/debian/openwrt/dl/clang-llvmorg-16-init-17653-g39da55e8-2.tgz" -C "third_party/llvm-build/Release+Asserts" ; echo -e "llvmorg-16-init-17653-g39da55e8-2" > "third_party/llvm-build/Release+Asserts/cr_build_revision" ; popd ; )
~/openwrt/build_dir/target-aarch64_cortex-a72_musl/naiveproxy-111.0.5563.64-1/src ~/openwrt/feeds/kenzo/naiveproxy
~/openwrt/feeds/kenzo/naiveproxy
touch /home/debian/openwrt/build_dir/target-aarch64_cortex-a72_musl/naiveproxy-111.0.5563.64-1/.prepared_00a9029f8b59ffea7ad6256ad07384c5_6664517399ebbbc92a37c5bb081b5c53
rm -f /home/debian/openwrt/build_dir/target-aarch64_cortex-a72_musl/naiveproxy-111.0.5563.64-1/.configured_*
rm -f /home/debian/openwrt/staging_dir/target-aarch64_cortex-a72_musl/stamp/.naiveproxy_installed
( pushd "/home/debian/openwrt/build_dir/target-aarch64_cortex-a72_musl/naiveproxy-111.0.5563.64-1/src" ; . ../init_env.sh "aarch64" "cortex-a72" "" "/home/debian/openwrt/staging_dir/toolchain-aarch64_cortex-a72_gcc-11.2.0_musl" ; export naive_flags+=" ${naive_ccache_flags}" ; mkdir -p "out" ; gn gen "out/Release" --args="${naive_flags}" --script-executable="python3" ; popd ; )
~/openwrt/build_dir/target-aarch64_cortex-a72_musl/naiveproxy-111.0.5563.64-1/src ~/openwrt/feeds/kenzo/naiveproxy
bash: line 1: gn: command not found
~/openwrt/feeds/kenzo/naiveproxy
touch /home/debian/openwrt/build_dir/target-aarch64_cortex-a72_musl/naiveproxy-111.0.5563.64-1/.configured_68b329da9893e34099c7d8ad5cb9c940
rm -f /home/debian/openwrt/build_dir/target-aarch64_cortex-a72_musl/naiveproxy-111.0.5563.64-1/.built
touch /home/debian/openwrt/build_dir/target-aarch64_cortex-a72_musl/naiveproxy-111.0.5563.64-1/.built_check
ninja -C "/home/debian/openwrt/build_dir/target-aarch64_cortex-a72_musl/naiveproxy-111.0.5563.64-1/src/out/Release" naive
ninja: Entering directory `/home/debian/openwrt/build_dir/target-aarch64_cortex-a72_musl/naiveproxy-111.0.5563.64-1/src/out/Release'
ninja: fatal: chdir to '/home/debian/openwrt/build_dir/target-aarch64_cortex-a72_musl/naiveproxy-111.0.5563.64-1/src/out/Release' - No such file or directory
make[3]: *** [Makefile:119: /home/debian/openwrt/build_dir/target-aarch64_cortex-a72_musl/naiveproxy-111.0.5563.64-1/.built] Error 1
make[3]: Leaving directory '/home/debian/openwrt/feeds/kenzo/naiveproxy'
time: package/feeds/kenzo/naiveproxy/compile#0.64#2.46#2.70
    ERROR: package/feeds/kenzo/naiveproxy failed to build.
make[2]: *** [package/Makefile:116: package/feeds/kenzo/naiveproxy/compile] Error 1
make[2]: Leaving directory '/home/debian/openwrt'
make[1]: *** [package/Makefile:110: /home/debian/openwrt/staging_dir/target-aarch64_cortex-a72_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/debian/openwrt'
make: *** [/home/debian/openwrt/include/toplevel.mk:230: world] Error 2
```

kenzok8, 2023-04-01:
缺少gn

edhnt455, 2023-04-01:
同样问题，插个眼，静等解决

edhnt455, 2023-04-02:
请问具体该如何解决呢

myannnascar, 2023-04-02:
> 请问具体该如何解决呢

`cd package`
`svn co https://github.com/kenzok8/small-package/trunk/gn`
暂时先把此工具下载在package目录，等后续作者将此添加到这个仓库吧

edhnt455, 2023-04-02:
> > 请问具体该如何解决呢
> 
> `cd package` `svn co https://github.com/kenzok8/small-package/trunk/gn` 暂时先把此工具下载在package目录，等后续作者将此添加到这个仓库吧

好的，谢谢