kenzok8/openwrt-packages #386: update 2023-04-09 02:08:52 会导致mips构架无法编译

SpeedPartner, 2023-04-09:
https://github.com/kenzok8/openwrt-packages/commit/a3e614486a8997edc6be6887a22accb5258d0d02
MT7621/mipsel_24kc_musl

日志:
```
Finding targets

Building targets
runtime/cgo
# runtime/cgo
gcc_mipsx.S: Assembler messages:
gcc_mipsx.S:25: Error: invalid operands `sw $18,12($29)'
gcc_mipsx.S:26: Error: invalid operands `sw $19,16($29)'
gcc_mipsx.S:27: Error: invalid operands `sw $20,20($29)'
gcc_mipsx.S:28: Error: invalid operands `sw $21,24($29)'
gcc_mipsx.S:29: Error: invalid operands `sw $22,28($29)'
gcc_mipsx.S:30: Error: invalid operands `sw $23,32($29)'
gcc_mipsx.S:31: Error: invalid operands `sw $30,36($29)'
gcc_mipsx.S:44: Error: invalid operands `jalr $20'
gcc_mipsx.S:48: Error: invalid operands `lw $18,12($29)'
gcc_mipsx.S:49: Error: invalid operands `lw $19,16($29)'
gcc_mipsx.S:50: Error: invalid operands `lw $20,20($29)'
gcc_mipsx.S:51: Error: invalid operands `lw $21,24($29)'
gcc_mipsx.S:52: Error: invalid operands `lw $22,28($29)'
gcc_mipsx.S:53: Error: invalid operands `lw $23,32($29)'
gcc_mipsx.S:54: Error: invalid operands `lw $30,36($29)'
gcc_mipsx.S:63: Error: invalid operands `lw $31,0($29)'

make[3]: *** [Makefile:72: /home/biutone/lede17/build_dir/target-mipsel_24kc_musl/alist-3.15.1/.built] Error 2
make[3]: Leaving directory '/home/biutone/lede17/feeds/kenzo/alist'
time: package/feeds/kenzo/alist/compile#2.28#0.75#3.05
make[2]: *** [package/Makefile:113: package/feeds/kenzo/alist/compile] Error 2
make[2]: Leaving directory '/home/biutone/lede17'
make[1]: *** [package/Makefile:107: /home/biutone/lede17/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/biutone/lede17'
make: *** [/home/biutone/lede17/include/toplevel.mk:227: world] Error 2
```

将`PKG_BUILD_FLAGS:=no-mips16`改回`PKG_USE_MIPS16:=0`后编译正常.
另外small里的[`update 2023-04-09 02:04:44`](https://github.com/kenzok8/small/commit/10cb3797afb63c139682cb9a3dca87bfeb29a9de)也是一样...

kenzok8, 2023-04-09:
https://github.com/kenzok8/openwrt-packages/commit/9bf5cd423564b0a3be662ac44f7109cbb6dbe2d6

SpeedPartner, 2023-04-09:
辛苦K大~

closed 2023-04-09 by SpeedPartner