kenzok8/openwrt-packages #387: 编译luci-app-gost的Go环境是怎样的？到目前没成功过

szdosar, 2023-04-12:
我的环境：
curl -Lo go.tar.gz https://go.dev/dl/go1.20.linux-amd64.tar.gz
sudo rm -rf /usr/local/go
sudo tar -C /usr/local -xzf go.tar.gz
rm go.tar.gz
ln -s /usr/local/go/bin/go /usr/bin/go
go install github.com/quic-go/qtls-go1-20@latest
go install github.com/onsi/gomega@latest
go install github.com/quic-go/qpack@latest
go install github.com/golang/mock@latest
go install github.com/francoispqt/gojay@latest
go install github.com/cheekybits/genny@latest
报错：
Building targets
github.com/lucas-clemente/quic-go/internal/qtls
# github.com/lucas-clemente/quic-go/internal/qtls
../../../../../dl/go-mod-cache/github.com/lucas-clemente/quic-go@v0.28.1/internal/qtls/go120.go:6:13: cannot use "The version of quic-go you're using can't be built on Go 1.20 yet. For more details, please see https://github.com/lucas-clemente/quic-go/wiki/quic-go-and-Go-versions." (untyped string constant "The version of quic-go you're using can't be built on Go 1.20 yet. F...) as int value in variable declaration

SpeedPartner, 2023-04-14:
很显然,你`src-git packages ...`源中的golang版本1.20太新了,在1.20上还无法编译.
可以删除`feeds/packages/lang/golang`,然后替换为老版本.

jakaz1280, 2023-04-25:
gost里的quic只支持1.16 1.17和1.18的go，lede里的go是1.20.2。

SpeedPartner, 2023-04-25:
其实2月4号就解决了,此issue其实可以关闭

closed 2023-04-26 by kenzok8