kenzok8/openwrt-packages #388: 在自定义软件源添加https://op.dllkids.xyz刷新列表出错，请问该如何解决，感谢。

rianjskis, 2023-04-13:
在自定义软件源中填入： src/gz openwrt_dllkids https://op.dllkids.xyz/packages/aarch64_generic/
刷新软件包列表提示：Collected errors:
 * opkg_download: Failed to download https://op.dllkids.xyz/packages/aarch64_generic//Packages.sig, wget returned 8.

SpeedPartner, 2023-04-14:
.....这都什么乱七八糟的issue

rianjskis, 2023-04-15:
> .....这都什么乱七八糟的issue 这个只能算下载站,不能用于添加到软件源

感谢解惑，因为刚编译完我在软件包中发现添加进来的软件源改成了:
src/gz openwrt_kenzo https://mirrors.cloud.tencent.com/lede/snapshots/packages/aarch64_generic/kenzo
src/gz openwrt_small https://mirrors.cloud.tencent.com/lede/snapshots/packages/aarch64_generic/small
刷新列表就会报错,网上没看到相似教程,所以来问问.
我随后在自定义软件源中添加
 src/gz kenzo https://github.com/kenzok8/openwrt-packages  
也还是不行,请问我该如何写,才能把编译中的这2个软件源添加进去并能使用.

SpeedPartner, 2023-04-15:
为啥不跟原来可用的软件源对比下呢？
比如`https://downloads.openwrt.org/releases/packages-22.03/aarch64_generic/packages/`
发现区别了吗？

- 它的目录下的都是编译好的ipk，不是编译用的源码
- 它有标准的index索引
- 它的目录下前5个文件：Packages、Packages.asc、Packages.gz、Packages.manifest、Packages.sig；目录下任何一个软件包更新后它们都要重新计算生成，这些文件包含了目录下每个软件包的信息和签名，使之成为openwrt**可信**软件源的必要部分

rianjskis, 2023-04-15:
> 为啥不跟原来可用的软件源对比下呢？ 比如`https://downloads.openwrt.org/releases/packages-22.03/aarch64_generic/packages/` 发现区别了吗？
>
>     * 它的目录下的都是编译好的ipk，不是编译用的源码
>
>     * 它有标准的index索引
>
>     * [...]

感谢解答，如您所说,我查看了编译好后自带的软件源地区如您所说包含了这5个文件才能算可用软件源,
如果我需要更新kenzok8大叔的软件包只能重新编译或者在https://op.dllkids.xyz中下载安装了,

closed 2023-04-15 by rianjskis

1372831265, 2024-04-21:
使用软件源请删除 opkg配置(/etc/opkg.conf) 中的 option check_signature

luochen1990, 2024-06-12:
其实可以搞一下, 就是用 gpg 签个名,  可能要在构建流程里加一些步骤