kenzok8/openwrt-packages #396: luci-theme-argone能不能还原

Eay5, 2023-05-07:
luci-theme-argone更新之后，太丑了，完全变了！能不能还原

kenzok8, 2023-05-08:
截图看看

Eay5, 2023-05-08:
是这样的,你这边的luci-theme-argon与https://github.com/jerrykuku/luci-theme-argon的不一样!最近你升级了,导致编译出的固件用以前的备份还原,系统就不能访问了.然后登录页界面也变了.

Eay5, 2023-05-08:
谢谢你的源,我已经把那个主题给替换为了 https://github.com/jerrykuku/luci-theme-argon

hatbhtbh, 2023-05-08:
> luci-theme-argone更新之后，太丑了，完全变了！能不能还原

之前的主题名称应该是：luci-theme-argonne替换了成了luci-theme-argone
现在的主题进入web管理，系统，管理权下有小图标显示不出的bug，需要修复或者还使用以前的主题
截图怎么上传不上来呢

hatbhtbh, 2023-05-08:
请保留以前的主题风格，感谢！

hatbhtbh, 2023-05-08:
> 谢谢你的来源，我已经把那个主题给替换为了https://github.com/jerrykuku/luci-theme-argon

请问你是怎么替换的，请指点一下，谢谢！

hatbhtbh, 2023-05-08:
之前的主题名称是：luci-theme-argonne替换了成了luci-theme-argone
现在的主题进入web管理，系统，管理权下有小图标显示不出的bug，
需要修复或者还使用以前的主题

[image]

kenzok8, 2023-05-08:
嗯，看到了，是bug，空了修复下

Eay5, 2023-05-08:
> > 谢谢你的来源，我已经把那个主题给替换为了https://github.com/jerrykuku/luci-theme-argon
> 
> 请问你是怎么替换的，请指点一下，谢谢！

git clone -b 18.06 https://github.com/jerrykuku/luci-theme-argon.git package/luci-theme-argonenew
在依赖里面加
然后主题改为luci-theme-argonenew,就好

hatbhtbh, 2023-05-10:
> 嗯，看到了，是bug，空了修复下

大神，你昨天修复的是好了，又出现其它的问题，主菜单小图标出现问题，你看截图
[image]
[image]

hatbhtbh, 2023-05-10:
> > 嗯，看到了，是bug，空了修复下
> 
> 大神，你昨天修复的是好了，又出现它的问题，主菜单小图标出现问题，你看截图 [image] [image]

hatbhtbh, 2023-05-10:
> > > 嗯，看到了，是bug，空了修复下
> > 
> > 
> > 大神，你昨天修复的很好了，又出现其它的问题，主菜单小图标出现问题，你看截图[image] [image]

hatbhtbh, 2023-05-11:
> https://github.com/kenzok8/compile-package/releases 1 删除错误主题
> 
> opkg 信息 luci-theme* opkg 删除 luci-theme-argone
> 
> 2 openwrt主页系统下剩余的无效主题名称 nano /etc/config/luci
> 
> 3 /usr/lib/lua/luci/view/themes/ 目录下删除失效包

我今天再次用L大源码本地编译了一下，主题恢复如初，感谢~~~~~

closed 2023-05-12 by kenzok8

SpeedPartner, 2023-05-15:
看标题进来前,我还以为是我把18.06的改炸了...🤣