kenzok8/openwrt-packages #411: lede源加这个软件源编译出错，什么原因求解？

magi-Z, 2023-07-31:
# github.com/winfsp/cgofuse/fuse
../../../../../dl/go-mod-cache/github.com/winfsp/cgofuse@v1.5.0/fuse/host_cgo.go:116:10: fatal error: fuse.h: No such file or directory
  116 | #include <fuse.h>
      |          ^~~~~~~~
compilation terminated.
github.com/alist-org/alist/v3/pkg/chanio
github.com/alist-org/alist/v3/pkg/gowebdav/cmd/gowebdav
gorm.io/driver/sqlite
github.com/alist-org/alist/v3/internal/bootstrap
github.com/alist-org/alist/v3/cmd
github.com/alist-org/alist/v3

make[3]: *** [Makefile:76: /home/magi/lede/build_dir/target-aarch64_generic_musl/alist-3.24.0/.built] Error 1
make[3]: Leaving directory '/home/magi/lede/feeds/kenzo/alist'
time: package/feeds/kenzo/alist/compile#112.38#20.08#105.28
    ERROR: package/feeds/kenzo/alist failed to build.
make[2]: *** [package/Makefile:116: package/feeds/kenzo/alist/compile] Error 1
make[2]: Leaving directory '/home/magi/lede'
make[1]: *** [package/Makefile:110: /home/magi/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/magi/lede'
make: *** [/home/magi/lede/include/toplevel.mk:231：world] 错误 2

kenzok8, 2023-07-31:
alist编译出错了alist/compile Error 1 写的很明白

magi-Z, 2023-07-31:
> alist编译出错了alist/compile Error 1如何解决：小白求教！多次编译总是这个出错！什么原因请指教

kenzok8, 2023-07-31:
原因很多，你可以先试着不去勾选alist与luci-app-alist，跑一次成功编译，如果本地（可以考虑把成功编译的dl文件家复制出来，防止出错后替换），然后再添加alist与luci-app-alist

kenzok8, 2023-07-31:
另外记得git pull && ./scripts/feeds update -a && ./scripts/feeds install -a 常规操作更新下，可能你没拉取最新的

magi-Z, 2023-07-31:
> 另外记得git pull && ./scripts/feeds update -a && ./scripts/feeds install -a 常规操作更新下，可能你没拉取最新的

> 另外记得git pull && ./scripts/feeds update -a && ./scripts/feeds install -a 常规操作更新下，可能你没拉取最新的

> 原因很多，你可以先试着不去勾选alist与luci-app-alist，跑一次成功编译，如果本地（可以考虑把成功编译的dl文件家复制出来，防止出错后替换），然后再添加alist与luci-app-alist

取消alist怎么还出错！这是什么鬼！
gcc-11.3.0_musl/usr/lib -L/home/magi/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/lib -Wl,-z,now -Wl,-z,relro'"  
Finding targets

Building targets
github.com/winfsp/cgofuse/fuse
# github.com/winfsp/cgofuse/fuse
../../../../../dl/go-mod-cache/github.com/winfsp/cgofuse@v1.5.0/fuse/host_cgo.go:116:10: fatal error: fuse.h: No such file or directory
  116 | #include <fuse.h>
      |          ^~~~~~~~
compilation terminated.

make[3]: *** [Makefile:76: /home/magi/lede/build_dir/target-aarch64_generic_musl/alist-3.24.0/.built] Error 1
make[3]: Leaving directory '/home/magi/lede/feeds/kenzo/alist'
time: package/feeds/kenzo/alist/compile#3.07#2.32#6.67
    ERROR: package/feeds/kenzo/alist failed to build.
make[2]: *** [package/Makefile:116: package/feeds/kenzo/alist/compile] Error 1
make[2]: Leaving directory '/home/magi/lede'
make[1]: *** [package/Makefile:110: /home/magi/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/magi/lede'
make: *** [/home/magi/lede/include/toplevel.mk:231：world] 错误 2

lrgcx, 2023-07-31:
我这两天也一直是alist报错。

kenzok8, 2023-07-31:
编译过程里https://mirror2.openwrt.org/sources/alist-3.24.0.tar.gz 下载404了，

lrgcx, 2023-07-31:
先执行make distclean，再make download -j8没报错了 现在正在编译。

legon752, 2023-08-03:
我今天编译也碰到勾选alist报错，重新编译去掉alist后成功编译

1ZUMIKun, 2023-08-27:
alist依赖fuse，可以sudo apt install libfuse-dev

closed 2023-09-29 by kenzok8