kenzok8/openwrt-packages #414: 大佬能不能更新下主题，现在的istore 没图标、显示错位

SterlingCarter, 2023-08-03:
请大佬更新下主题，现在的istore 没图标、显示错位。我看留言里面很多网友都曾反映过这个问题，还请大佬能够更新一下主题适配一下Istore插件，谢谢啦
[image]
[image]

TheLastManT1, 2023-12-14:
直到今年都没有修复

jianxtop, 2023-12-21:
他们只是仓库搬运工，咋个修复？？我也打算自己整一个同步任务的仓库，自己维护版本相对稳妥一些

TheLastManT1, 2023-12-22:
> 他们只是仓库搬运工，咋个修复？？我也打算自己整一个同步任务的仓库，自己维护版本相对稳妥一些

好多主题都有这个bug

kenzok8, 2026-02-24:
自动巡检：该 Issue 长时间未更新（>=180 天），先关闭以便集中处理当前问题。

如问题仍存在，请按最新源码重新复现后提交新 Issue，并附：
- 固件版本与目标架构
- 完整编译/运行日志（从第一个报错开始）
- 复现步骤

Auto triage: this issue has been inactive for >=180 days, so it is being closed for backlog cleanup.
If the issue still exists, please open a new issue with updated logs and clear reproduction steps.