kenzok8/openwrt-packages #416: Xray-core-1.8.4 编译错误，用的大雕lede源

dukmonki, 2023-09-01:
make[3]: Leaving directory '/home/lede/lede/feeds/packages/lang/golang/golang'
time: package/feeds/packages/golang/host-compile#0.22#0.75#0.76
make[3]: Entering directory '/home/lede/lede/feeds/small8/xray-core'
rm -f /home/lede/lede/build_dir/target-mipsel_24kc_musl/Xray-core-1.8.4/.built
touch /home/lede/lede/build_dir/target-mipsel_24kc_musl/Xray-core-1.8.4/.built_check
CONFIG_GOLANG_MOD_CACHE_WORLD_READABLE="" GO_BUILD_CACHE_DIR="/home/lede/lede/tmp/go-build" GO_MOD_CACHE_DIR="/home/lede/lede/dl/go-mod-cache" GO_MOD_ARGS="-modcacherw" GO_PKG="github.com/xtls/xray-core" GO_INSTALL_EXTRA="" GO_INSTALL_ALL="" GO_SOURCE_ONLY="" GO_BUILD_PKG="github.com/xtls/xray-core/main" GO_EXCLUDES="" GO_GO_GENERATE="" GO_INSTALL_BIN_PATH="/usr/bin" BUILD_DIR="/home/lede/lede/build_dir/target-mipsel_24kc_musl/Xray-core-1.8.4" GO_BUILD_DIR="/home/lede/lede/build_dir/target-mipsel_24kc_musl/Xray-core-1.8.4/.go_work/build" GO_BUILD_BIN_DIR="/home/lede/lede/build_dir/target-mipsel_24kc_musl/Xray-core-1.8.4/.go_work/build/bin/linux_mipsle" GO_BUILD_DEPENDS_PATH="/usr/share/gocode" GO_BUILD_DEPENDS_SRC="/home/lede/lede/staging_dir/target-mipsel_24kc_musl/usr/share/gocode/src" GOOS="linux" GOARCH="mipsle" GO386="" GOAMD64="" GOARM="" GOMIPS="softfloat" GOMIPS64="" GOPPC64="" CGO_ENABLED=1 CC="mipsel-openwrt-linux-musl-gcc" CXX="mipsel-openwrt-linux-musl-g++" CGO_CFLAGS="-Os -pipe -mno-branch-likely -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -msoft-float -mips16 -minterlink-mips16 -fmacro-prefix-map=/home/lede/lede/build_dir/target-mipsel_24kc_musl/Xray-core-1.8.4=Xray-core-1.8.4 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_CPPFLAGS="-I/home/lede/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/usr/include -I/home/lede/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/include/fortify -I/home/lede/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/include" CGO_CXXFLAGS="-Os -pipe -mno-branch-likely -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -msoft-float -mips16 -minterlink-mips16 -fmacro-prefix-map=/home/lede/lede/build_dir/target-mipsel_24kc_musl/Xray-core-1.8.4=Xray-core-1.8.4 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_LDFLAGS="-L/home/lede/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/usr/lib -L/home/lede/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/lib -znow -zrelro" GOPATH="/home/lede/lede/build_dir/target-mipsel_24kc_musl/Xray-core-1.8.4/.go_work/build" GOCACHE="/home/lede/lede/tmp/go-build" GOMODCACHE="/home/lede/lede/dl/go-mod-cache" GOENV=off /usr/bin/env bash /home/lede/lede/feeds/packages/lang/golang//golang-build.sh build -v -buildvcs=false -trimpath -ldflags "all=-buildid '1693466411' -linkmode external -extldflags '-L/home/lede/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/usr/lib -L/home/lede/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/lib -Wl,-z,now -Wl,-z,relro'"     -installsuffix "softfloat"   -ldflags " -X github.com/xtls/xray-core/core.build=OpenWrt -X github.com/xtls/xray-core/core.version=1.8.4 -buildid '1693466411' -linkmode external -extldflags '-L/home/lede/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/usr/lib -L/home/lede/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/lib -Wl,-z,now -Wl,-z,relro'"  
Finding targets

Building targets
runtime/cgo
# runtime/cgo
gcc_mipsx.S: Assembler messages:
gcc_mipsx.S:27: Error: invalid operands `sw $18,12($29)'
gcc_mipsx.S:28: Error: invalid operands `sw $19,16($29)'
gcc_mipsx.S:29: Error: invalid operands `sw $20,20($29)'
gcc_mipsx.S:30: Error: invalid operands `sw $21,24($29)'
gcc_mipsx.S:31: Error: invalid operands `sw $22,28($29)'
gcc_mipsx.S:32: Error: invalid operands `sw $23,32($29)'
gcc_mipsx.S:33: Error: invalid operands `sw $30,36($29)'
gcc_mipsx.S:46: Error: invalid operands `jalr $20'
gcc_mipsx.S:50: Error: invalid operands `lw $18,12($29)'
gcc_mipsx.S:51: Error: invalid operands `lw $19,16($29)'
gcc_mipsx.S:52: Error: invalid operands `lw $20,20($29)'
gcc_mipsx.S:53: Error: invalid operands `lw $21,24($29)'
gcc_mipsx.S:54: Error: invalid operands `lw $22,28($29)'
gcc_mipsx.S:55: Error: invalid operands `lw $23,32($29)'
gcc_mipsx.S:56: Error: invalid operands `lw $30,36($29)'
gcc_mipsx.S:65: Error: invalid operands `lw $31,0($29)'

Makefile:88: recipe for target '/home/lede/lede/build_dir/target-mipsel_24kc_musl/Xray-core-1.8.4/.built' failed
make[3]: *** [/home/lede/lede/build_dir/target-mipsel_24kc_musl/Xray-core-1.8.4/.built] Error 1
make[3]: Leaving directory '/home/lede/lede/feeds/small8/xray-core'
time: package/feeds/small8/xray-core/compile#0.53#1.02#1.12
    ERROR: package/feeds/small8/xray-core failed to build.
package/Makefile:114: recipe for target 'package/feeds/small8/xray-core/compile' failed
make[2]: *** [package/feeds/small8/xray-core/compile] Error 1
make[2]: Leaving directory '/home/lede/lede'
package/Makefile:110: recipe for target '/home/lede/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/home/lede/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/lede/lede'
/home/lede/lede/include/toplevel.mk:229: recipe for target 'world' failed
make: *** [world] Error 2

jvesky, 2023-09-02:
先说结论：就是 kenzok8 这边的仓库有问题，希望大佬尽快处理，如需SSR+，请先用 https://github.com/fw876/helloworld 的。

这个问题我前天就想提交了，本地和云编译，反复测试的结果如下。

测试环境，本地和云编译均为 Ubuntu 20.04 LTS，采用 Lede 源码。

一开始我是本地编译，一直出错，怀疑可能是梯子或者编译环境有问题，后来做了如下测试。
1，本地、云端拉取 kenzok8/openwrt-packages 和 kenzok8/small 源，本地和云端均编译出错。
2、本地、远端拉取 kenzok8/small 源，本地和云端均编译出错。

出错内容均为 xray-core 编译失败，如下。

`make[3]: *** [Makefile:91: /workdir/openwrt/build_dir/target-mipsel_24kc_musl/Xray-core-1.8.4/.built] Error 1
make[3]: Leaving directory '/workdir/openwrt/feeds/small/xray-core'
time: package/feeds/small/xray-core/compile#1.20#0.32#1.08
    ERROR: package/feeds/small/xray-core failed to build.
make[2]: *** [package/Makefile:116: package/feeds/small/xray-core/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.`

3、没有添加何第三方源，为了排除是不是其他插件影响，插件只选择了”防火墙“和”SSR+“（SSR+ 通过 git clone https://github.com/fw876/helloworld.git feeds/luci/applications/helloworld 获取） 编译成功。
4、跟第三条一样，但是添加了其他的常用插件，编译成功。

SpeedPartner, 2023-09-02:
老问题了,https://github.com/kenzok8/small/commit/668304011bfa733b6a1a507b4f8f7da7d8398af3
只要用了`PKG_BUILD_FLAGS:=no-mips16`必定导致mips平台无法编译

closed 2023-10-10 by kenzok8