kenzok8/openwrt-packages #417: Github Action 云编译 Alist总是出错，本地编译又是OK的，请帮忙看看。

Azreulo, 2023-09-07:
同样的.config文件，安装同样的依赖，在本地编译就可以成功，云编译alist总是出错，日志如下，貌似还是golang版本的问题，求指点如何修正。

已安装libfuse-dev。

```
2023-09-07T08:53:05.1832387Z github.com/winfsp/cgofuse/fuse
2023-09-07T08:53:05.2548913Z # github.com/winfsp/cgofuse/fuse
2023-09-07T08:53:05.2550422Z ../../../../../dl/go-mod-cache/github.com/winfsp/cgofuse@v1.5.1-0.20230130140708-f87f5db493b5/fuse/host_cgo.go:117:10: fatal error: fuse.h: No such file or directory
2023-09-07T08:53:05.2551127Z   117 | #include <fuse.h>
2023-09-07T08:53:05.2551507Z       |          ^~~~~~~~
2023-09-07T08:53:05.2552022Z compilation terminated.
2023-09-07T08:53:05.2552678Z github.com/alist-org/alist/v3/pkg/chanio
2023-09-07T08:53:05.2718269Z github.com/alist-org/alist/v3/pkg/gowebdav/cmd/gowebdav
2023-09-07T08:53:23.8744132Z gorm.io/driver/sqlite
2023-09-07T08:53:24.0311169Z github.com/alist-org/alist/v3/internal/bootstrap
2023-09-07T08:53:24.1148728Z github.com/alist-org/alist/v3/cmd
2023-09-07T08:53:24.3198644Z github.com/alist-org/alist/v3
2023-09-07T08:53:26.8140333Z 
2023-09-07T08:53:26.9287261Z make[3]: *** [Makefile:76: /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/target-x86_64_musl/alist-3.27.0/.built] Error 1
2023-09-07T08:53:26.9291116Z make[3]: Leaving directory '/home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/feeds/kenzok8/alist'
2023-09-07T08:53:26.9291961Z time: package/feeds/kenzok8/alist/compile#192.66#21.78#129.90
2023-09-07T08:53:26.9298390Z     ERROR: package/feeds/kenzok8/alist failed to build.
2023-09-07T08:53:26.9303453Z make[2]: *** [package/Makefile:116: package/feeds/kenzok8/alist/compile] Error 1
2023-09-07T08:53:26.9310377Z make[2]: Leaving directory '/home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt'
2023-09-07T08:53:26.9324599Z make[1]: *** [package/Makefile:110: /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
2023-09-07T08:53:26.9326051Z make[1]: Leaving directory '/home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt'
2023-09-07T08:53:26.9338477Z make: *** [/home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/include/toplevel.mk:231: world] Error 2
```

kenzok8, 2023-09-07:
1、把本地最新能编译通过的config文件上传或替换
2、另外比如云编译里添加sudo apt install libfuse-dev
3、云编译试试添加命令 rm -rf feeds/packages/lang/golang
git clone https://github.com/sbwml/packages_lang_golang -b 20.x feeds/packages/lang/golang

Azreulo, 2023-09-08:
Thanks. 问题解决了…然后，我发现apt后面少了install没发现。我是菜鸡。

closed 2023-09-08 by Azreulo