kenzok8/openwrt-packages #424: alist错误

qq1316237299, 2023-10-15:
Makefile:73: recipe for target '/home/qq1316237299/lede/build_dir/target-x86_64_musl/alist-3.28.0/.built' failed
make[3]: *** [/home/qq1316237299/lede/build_dir/target-x86_64_musl/alist-3.28.0/.built] Error 1
make[3]: Leaving directory '/home/qq1316237299/lede/feeds/kenzo/alist'
time: package/feeds/kenzo/alist/compile#1.31#1.54#1.65
    ERROR: package/feeds/kenzo/alist failed to build.
package/Makefile:114: recipe for target 'package/feeds/kenzo/alist/compile' failed
make[2]: *** [package/feeds/kenzo/alist/compile] Error 1
make[2]: Leaving directory '/home/qq1316237299/lede'
package/Makefile:110: recipe for target '/home/qq1316237299/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/qq1316237299/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/qq1316237299/lede'
/home/qq1316237299/lede/include/toplevel.mk:229: recipe for target 'world' failed
[image]

kenzok8, 2023-10-15:
sudo apt install libfuse-dev

qq1316237299, 2023-10-15:
> sudo apt install libfuse-dev

我试试看，谢谢大佬

qq1316237299, 2023-10-15:
> sudo apt install libfuse-dev

可以啦  谢谢

closed 2023-10-15 by qq1316237299