kenzok8/openwrt-packages #426: 哪位大佬帮忙看一下这个问题怎么解决？

hwebuse, 2023-10-22:
Collected errors:
 * check_data_file_clashes: Package libnetwork wants to install file /home/yk/lede/build_dir/target-x86_64_musl/root-x86/usr/bin/docker-proxy
        But that file is already provided by package  * dockerd
 * opkg_install_cmd: Cannot install package libnetwork.
make[2]: *** [package/Makefile:70: package/install] Error 255
make[2]: Leaving directory '/home/yk/lede'
make[1]: *** [package/Makefile:111: /home/yk/lede/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/yk/lede'
make: *** [/home/yk/lede/include/toplevel.mk:231：world] 错误 2

10月18号还能正常编译，同样的配置，22号就不能编译了。。

qzldd, 2023-10-23:
出现了同样的问题，等大佬

Zane-E, 2023-10-23:
https://github.com/coolsnowwolf/lede/issues/11611

closed 2023-11-09 by kenzok8