kenzok8/openwrt-packages #427: adguardhome：报错

ywt114, 2023-10-23:
2023-10-23T14:44:02.0490383Z make[3]: Entering directory '/home/runner/work/OpenWrt/OpenWrt/openwrt/feeds/packages/net/adguardhome'
2023-10-23T14:44:02.2218605Z touch /home/runner/work/OpenWrt/OpenWrt/openwrt/build_dir/target-x86_64_musl/adguardhome-0.107.40/.prepared_e470ef84a780eb9e2a10be28bf483bc3_6664517399ebbbc92a37c5bb081b5c53_check
2023-10-23T14:44:02.2228465Z . /home/runner/work/OpenWrt/OpenWrt/openwrt/include/shell.sh; /home/runner/work/OpenWrt/OpenWrt/openwrt/staging_dir/host/bin/libdeflate-gzip -dc /home/runner/work/OpenWrt/OpenWrt/openwrt/dl/adguardhome-0.107.40.tar.gz | tar -C /home/runner/work/OpenWrt/OpenWrt/openwrt/build_dir/target-x86_64_musl/adguardhome-0.107.40/.. -xf -
2023-10-23T14:44:02.3003887Z [ ! -d ./src/ ] || cp -fpR ./src/. /home/runner/work/OpenWrt/OpenWrt/openwrt/build_dir/target-x86_64_musl/adguardhome-0.107.40
2023-10-23T14:44:02.3071573Z 
2023-10-23T14:44:02.3075220Z Applying ./patches/010-go-mod-tidy.patch using plaintext: 
2023-10-23T14:44:02.3087111Z can't find file to patch at input line 3
2023-10-23T14:44:02.3087937Z Perhaps you used the wrong -p or --strip option?
2023-10-23T14:44:02.3088607Z The text leading up to this was:
2023-10-23T14:44:02.3089165Z --------------------------
2023-10-23T14:44:02.3089637Z |--- a/go.mod
2023-10-23T14:44:02.3090029Z |+++ b/go.mod
2023-10-23T14:44:02.3090456Z --------------------------
2023-10-23T14:44:02.3090968Z No file to patch.  Skipping patch.
2023-10-23T14:44:02.3091512Z 2 out of 2 hunks ignored
2023-10-23T14:44:02.3092079Z can't find file to patch at input line 23
2023-10-23T14:44:02.3092814Z Perhaps you used the wrong -p or --strip option?
2023-10-23T14:44:02.3093774Z The text leading up to this was:
2023-10-23T14:44:02.3094282Z --------------------------
2023-10-23T14:44:02.3097564Z |--- a/go.sum
2023-10-23T14:44:02.3100298Z make[3]: *** [Makefile:88: /home/runner/work/OpenWrt/OpenWrt/openwrt/build_dir/target-x86_64_musl/adguardhome-0.107.40/.prepared_e470ef84a780eb9e2a10be28bf483bc3_6664517399ebbbc92a37c5bb081b5c53] Error 1
2023-10-23T14:44:02.3102014Z |+++ b/go.sum
2023-10-23T14:44:02.3108575Z --------------------------
2023-10-23T14:44:02.3109119Z No file to patch.  Skipping patch.
2023-10-23T14:44:02.3109674Z 8 out of 8 hunks ignored
2023-10-23T14:44:02.3110457Z Patch failed!  Please fix ./patches/010-go-mod-tidy.patch!
2023-10-23T14:44:02.3111734Z make[3]: Leaving directory '/home/runner/work/OpenWrt/OpenWrt/openwrt/feeds/packages/net/adguardhome'
2023-10-23T14:44:02.3112972Z time: package/feeds/packages/adguardhome/compile#0.18#0.09#0.27
2023-10-23T14:44:02.3113868Z     ERROR: package/feeds/packages/adguardhome failed to build.
2023-10-23T14:44:02.3123786Z make[2]: *** [package/Makefile:116: package/feeds/packages/adguardhome/compile] Error 1
2023-10-23T14:44:02.3125557Z make[2]: Leaving directory '/home/runner/work/OpenWrt/OpenWrt/openwrt'
2023-10-23T14:44:02.3128009Z make[1]: *** [package/Makefile:110: /home/runner/work/OpenWrt/OpenWrt/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
2023-10-23T14:44:02.3134712Z make[1]: Leaving directory '/home/runner/work/OpenWrt/OpenWrt/openwrt'
2023-10-23T14:44:02.3142367Z make: *** [/home/runner/work/OpenWrt/OpenWrt/openwrt/include/toplevel.mk:231: world] Error 2
2023-10-23T14:44:02.3204095Z ##[error]Process completed with exit code 2.

这个错误似乎与 https://github.com/kenzok8/openwrt-packages/blob/master/adguardhome/patches/010-go-mod-tidy.patch 补丁有关，提示无法找到要修补的文件。在处理 go.mod 和 go.sum 时出现了问题。只更新了luci没有更新对应依赖。

kenzok8, 2023-10-23:
删除补丁
2023-10-23T14:44:02.0490383Z make[3]: Entering directory '/home/runner/work/OpenWrt/OpenWrt/openwrt/feeds/packages/net/adguardhome'
2023-10-23T14:44:02.2218605Z touch /home/runner/work/OpenWrt/OpenWrt/openwrt/build_dir/target-x86_64_musl/adguardhome-0.107.40/.prepared_e470ef84a780eb9e2a10be28bf483bc3_6664517399ebbbc92a37c5bb081b5c53_check
2023-10-23T14:44:02.2228465Z . /home/runner/work/OpenWrt/OpenWrt/openwrt/include/shell.sh; /home/runner/work/OpenWrt/OpenWrt/openwrt/staging_dir/host/bin/libdeflate-gzip -dc /home/runner/work/OpenWrt/OpenWrt/openwrt/dl/adguardhome-0.107.40.tar.gz | tar -C /home/runner/work/OpenWrt/OpenWrt/openwrt/build_dir/target-x86_64_musl/adguardhome-0.107.40/.. -xf -
2023-10-23T14:44:02.3003887Z [ ! -d ./src/ ] || cp -fpR ./src/. /home/runner/work/OpenWrt/OpenWrt/openwrt/build_dir/target-x86_64_musl/adguardhome-0.107.40
2023-10-23T14:44:02.3071573Z
2023-10-23T14:44:02.3075220Z Applying ./patches/010-go-mod-tidy.patch using plaintext:
2023-10-23T14:44:02.3087111Z can't find file to patch at input line 3
2023-10-23T14:44:02.3087937Z Perhaps you used the wrong -p or --strip option?
2023-10-23T14:44:02.3088607Z The text leading up to this was:
2023-10-23T14:44:02.3089165Z --------------------------
2023-10-23T14:44:02.3089637Z |--- a/go.mod
2023-10-23T14:44:02.3090029Z |+++ b/go.mod
2023-10-23T14:44:02.3090456Z --------------------------
2023-10-23T14:44:02.3090968Z No file to patch.  Skipping patch.
2023-10-23T14:44:02.3091512Z 2 out of 2 hunks ignored
2023-10-23T14:44:02.3092079Z can't find file to patch at input line 23
2023-10-23T14:44:02.3092814Z Perhaps you used the wrong -p or --strip option?
2023-10-23T14:44:02.3093774Z The text leading up to this was:
2023-10-23T14:44:02.3094282Z --------------------------
2023-10-23T14:44:02.3097564Z |--- a/go.sum
2023-10-23T14:44:02.3100298Z make[3]: *** [Makefile:88: /home/runner/work/OpenWrt/OpenWrt/openwrt/build_dir/target-x86_64_musl/adguardhome-0.107.40/.prepared_e470ef84a780eb9e2a10be28bf483bc3_6664517399ebbbc92a37c5bb081b5c53] Error 1
2023-10-23T14:44:02.3102014Z |+++ b/go.sum
2023-10-23T14:44:02.3108575Z --------------------------
2023-10-23T14:44:02.3109119Z No file to patch.  Skipping patch.
2023-10-23T14:44:02.3109674Z 8 out of 8 hunks ignored
2023-10-23T14:44:02.3110457Z Patch failed!  Please fix ./patches/010-go-mod-tidy.patch!
2023-10-23T14:44:02.3111734Z make[3]: Leaving directory '/home/runner/work/OpenWrt/OpenWrt/openwrt/feeds/packages/net/adguardhome'
2023-10-23T14:44:02.3112972Z time: package/feeds/packages/adguardhome/compile#0.18#0.09#0.27
2023-10-23T14:44:02.3113868Z     ERROR: package/feeds/packages/adguardhome failed to build.
2023-10-23T14:44:02.3123786Z make[2]: *** [package/Makefile:116: package/feeds/packages/adguardhome/compile] Error 1
2023-10-23T14:44:02.3125557Z make[2]: Leaving directory '/home/runner/work/OpenWrt/OpenWrt/openwrt'
2023-10-23T14:44:02.3128009Z make[1]: *** [package/Makefile:110: /home/runner/work/OpenWrt/OpenWrt/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
2023-10-23T14:44:02.3134712Z make[1]: Leaving directory '/home/runner/work/OpenWrt/OpenWrt/openwrt'
2023-10-23T14:44:02.3142367Z make: *** [/home/runner/work/OpenWrt/OpenWrt/openwrt/include/toplevel.mk:231: world] Error 2
2023-10-23T14:44:02.3204095Z ##[error]Process completed with exit code 2.
这个错误似乎与 https://github.com/kenzok8/openwrt-packages/blob/master/adguardhome/patches/010-go-mod-tidy.patch 补丁有关，提示无法找到要修补的文件。在处理 go.mod 和 go.sum 时出现了问题。只更新了luci没有更新对应依赖。

ywt114, 2023-10-23:
@kenzok8 不是应该你从库里删除来修复吗

closed 2023-10-28 by kenzok8

fairok, 2023-11-17:
我的用新代码编译还是一样的错误呢

fairok, 2023-12-15:
> @kenzok8 不是应该你从库里删除来修复吗

楼主问题解决了吗？