kenzok8/openwrt-packages #433: luci-app-wechatpush/Makefile' has a dependency on 'wrtbwmon', which does not exist

ChaingTsung, 2023-11-11:
yoki@vps72371310:~/openwrt$ make menuconfig
Collecting package info: done
WARNING: Makefile 'package/feeds/kenzo/luci-app-wechatpush/Makefile' has a dependency on 'wrtbwmon', which does not exist

*** End of the configuration.
*** Execute 'make' to start the build or try 'make help'.

yoki@vps72371310:~/openwrt$ df -h
Filesystem      Size  Used Avail Use% Mounted on
tmpfs           299M  1.1M  297M   1% /run
/dev/vda1        49G  7.2G   39G  16% /
tmpfs           1.5G     0  1.5G   0% /dev/shm
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs           299M  4.0K  299M   1% /run/user/0
tmpfs           1.5G     0  1.5G   0% /run/qemu
yoki@vps72371310:~/openwrt$

tedaz, 2023-11-17:
same issue
`WARNING: Makefile 'package/feeds/kenzo/luci-app-wechatpush/Makefile' has a dependency on 'wrtbwmon', which does not exist
`

xjinhua, 2023-11-17:
我的也一样，这个报错

dudulongg, 2023-12-05:
关闭的issue里说是删库，但是不知道怎么删，不敢乱动，虽然是警告不影响编译，但是看着还是不太舒服，希望大佬们指点一下怎么处理这个警告

kenzok8, 2026-02-24:
自动巡检：该 Issue 长时间未更新（>=180 天），先关闭以便集中处理当前问题。

如问题仍存在，请按最新源码重新复现后提交新 Issue，并附：
- 固件版本与目标架构
- 完整编译/运行日志（从第一个报错开始）
- 复现步骤

Auto triage: this issue has been inactive for >=180 days, so it is being closed for backlog cleanup.
If the issue still exists, please open a new issue with updated logs and clear reproduction steps.