kenzok8/openwrt-packages #436: 使用design发生错误

bb955, 2023-11-21:
上传[design](https://op.dllkids.xyz/packages/aarch64_cortex-a53/luci-theme-design_new_git-24.308.56369-595cdca_all.ipk)修改为这个主题后，路由器页面发生错误。
[image]

请问如何急救

kenzok8, 2023-11-21:
1 删除错误主题
opkg info luci-theme*
opkg remove luci-theme-argone

2 openwrt系统下多余的无效主题名
nano /etc/config/luci

3 /usr/lib/lua/luci/view/themes/
目录下删除失效包

bb955, 2023-11-21:
OpenWrt 更改主题后无法进后台，主要原因是新主题与 OpenWrt 版本不兼容，导致 Luci 后台页面无法加载。解决方法如下：

1. **重启 OpenWrt**

重启 OpenWrt 可能会修复一些临时性错误，尝试一下。

2. **关闭 Luci 缓存**

Luci 使用缓存来加快页面加载速度，如果缓存中存储了有问题的主题信息，可能会导致无法加载后台页面。可以通过以下命令关闭 Luci 缓存：

```
uci set luci.ccache.enable=0
uci commit luci
```

3. **卸载新主题**

如果重启和关闭缓存都无法解决问题，可以尝试卸载新主题。可以通过以下命令卸载主题：

```
opkg remove luci-theme-[主题名称]
```

例如，要卸载 Argon 主题，可以使用以下命令：

```
opkg remove luci-theme-argon
```

卸载主题后，OpenWrt 将恢复默认主题。

4. **更新 OpenWrt 到最新版本**

如果您使用的是旧版本的 OpenWrt，可以尝试更新到最新版本。最新版本的 OpenWrt 通常会修复一些已知的兼容性问题。

5. **使用其他主题**

如果您确定新主题与 OpenWrt 版本兼容，但仍然无法加载后台页面，可以尝试使用其他主题。

您可以从 OpenWrt 官方仓库或第三方仓库下载主题。

以下是一些常用的主题：

* **默认主题**：OpenWrt 自带的主题，兼容性较好。
* **Aragon**：一款简洁现代的主题。
* **Material**：一款基于 Material Design 的主题。
* **Paper**：一款扁平设计的主题。

希望以上方法能帮助您解决问题。

closed 2026-02-24 by kenzok8