kenzok8/openwrt-packages #437: 使用lean源码加入您这边的AdGuardHome报错，缺少go.mod文件

fxxccp, 2023-11-25:
如题所示，编译5.15内核x86固件，加入AdGuardHome插件后编译过程中会报错，提示为缺少go.mod文件。
以下是详细日志：

make[3] -C package/adguardhome compile
ERROR: package/adguardhome failed to build.
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 1
make[2]: Entering directory '/workdir/openwrt/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/workdir/openwrt/scripts/config'
make[1] world
make[2] target/compile
make[3] -C target/linux compile
make[2] buildinfo
make[3] diffconfig buildversion feedsversion
make[4] scripts/config/conf
make[2] package/cleanup
make[2] package/compile
make[3] -C package/libs/libjson-c host-compile
make[3] -C package/libs/libubox host-compile
make[3] -C package/system/opkg host-compile
make[3] -C feeds/packages/lang/golang/golang host-compile
make[3] -C package/libs/toolchain compile
make[3] -C package/system/ca-certificates compile
make[3] -C package/adguardhome compile
ERROR: package/adguardhome failed to build.
……
……
……
no required module provides package github.com/AdguardTeam/AdGuardHome: go.mod file not found in current directory or any parent directory; see 'go help modules'

Building targets
go: cannot find main module, but found .git/config in /workdir/openwrt
to create a module there, run:
cd ../../../../.. && go mod init

make[3]: *** [Makefile:84: /workdir/openwrt/build_dir/target-x86_64_musl/adguardhome-0.107.41/.built] Error 1
make[3]: Leaving directory '/workdir/openwrt/package/adguardhome'
time: package/adguardhome/compile#0.23#0.06#0.28
ERROR: package/adguardhome failed to build.
make[2]: *** [package/Makefile:116: package/adguardhome/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.

fairok, 2023-12-05:
我也遇到了

smdx, 2024-01-07:
[image]

我这云编译也出现了这个问题

fairok, 2024-01-29:
楼主这个问题解决了吗？

closed 2024-03-16 by kenzok8