kenzok8/openwrt-packages #444: lean源码编译此仓库里的adguardhome报错找不到go main module

Winnow-Wit, 2024-01-04:
Lean源码编译kenzok8大佬仓库里的adguardhome，报错显示找不到go main module，查看build_dir发现此directory下是有go.mod go.sum和main.go的。这不是编译adguardhome的luci-app
**报错信息如下**：
_check
CONFIG_GOLANG_MOD_CACHE_WORLD_READABLE="" GO_BUILD_CACHE_DIR="/home/ubuntu/lede/tmp/go-build" GO_MOD_CACHE_DIR="/home/ubuntu/lede/dl/go-mod-cache"GO_MOD_ARGS="-modcacherw" GO_PKG="github.com/AdguardTeam/AdGuardHome" GO_INSTALL_EXTRA="!" GO_INSTALL_ALL="" GO_SOURCE_ONLY="" GO_BUILD_PKG="github.com/AdguardTeam/AdGuardHom e" GO_EXCLUDES="" GO_GO_GENERATE=""' GO_INSTALL_BIN_PATH="/usr/bin" BUILD_DIR="/h ome/ubuntu/lede/build_dir/target-x86_64_musl/adguardhome-0.107.43" GO_BUILD_DIR="/home/ubuntu/lede/build_dir/target-x86_64_musl/adguardhome-0.107.43/.go_work/bu ild" GO_BUILD_BIN_DIR="/home/ubuntu/lede/build_dir/target-x86_64_musl/adguardhom e-0.107.43/.go_work/build/bin" GO_BUILD_DEPENDS_PATH="/usr/share/gocode" GO_BUIL D_DEPENDS_SRC="/home/ubuntu/lede/staging_dir/target-x86_64_musl/usr/share/gocode/src" GOOS="Linux" GOARCH="amd64" G0386="" GOAMD64="v1" GOARM="" GOMIPS="" GOMIP S64="" GOPPC64="" CGO_ENABLED=1 CC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-0 penwrt-linux-musl-g++" CGO_CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/ubuntu/lede/build_dir/target-x86_64_musl/adguard home-0.107.43=adguardhome-0.107.43 -Wformat -Werror=format-security -fstack-prot ector -D_FORTIFY_SOURCE=1 -Wl,-z,now -wl,-z,relro" CGO_CPPFLAGS="-I/home/ubuntu/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/include -I/home/ubuntu/led e/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify -I/home/ubuntu/le de/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include" CGO_CXXFLAGS="-Os -pipe
-fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/ubuntu/lede/build_dir/target-x86_64_musl/adguardhome-0.107.43=adguardhome-0.107.43 -wformat-Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -wl,-z,now -wl,-z,relro" CGO_LDFLAGS="-L/home/ubuntu/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/lib -L/home/ubuntu/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/li b -znow -zrelro" GOPATH="/home/ubuntu/lede/build_dir/target-x86_64_musl/adguardh ome-0.107.43/.go_work/build" GOCACHE="/home/ubuntu/lede/tmp/go-build"GOMODCACHE="/home/ubuntu/lede/dl/go-mod-cache" GOENV=off GOTOOLCHAIN=local /usr/bin/env ba sh /home/ubuntu/lede/feeds/packages/lang/golang//golang-build.sh build -v -build vcs=false -trimpath -ldflags "all=-buildid '1703997625' -linkmode external -extl dflags '-L/home/ubuntu/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/lib-L/home/ubuntu/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib -wl,-z,now-Wl,-z,relro'"
-ldflags " -X github.com/AdguardTeam/AdGuardHome/internal/version.channel=release -X github.com/AdguardTeam/AdGuardHome/internal/version.version=v0.107.43 -X github.com/AdguardTeam/AdGuardHome/internal/version.buildti me=2023-12-31T04:40:25Z+0000 -X github.com/AdguardTeam/AdGuardHome/internal/vers ion.goarm= -X github.com/AdguardTeam/AdGuardHome/internal/version.gomips= -build id '1703997625'-linkmode external -extldflags '-L/home/ubuntu/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/lib -L/home/ubuntu/lede/staging_dir/toolcha in-x86_64_gcc-11.3.0_musl/lib -wl,-z,now -wl,-z,relro'"

**Finding targets**
**no required module provides package github.com/AdguardTeam/AdGuardHome:go.mod f
ile not found in current directory or any parent directory; see 'go help modules
Building targets
go: cannot find main module, but found .git/config in /home/ubuntu/lede
to create a module there, run:
cd ../..1..1..1.. && go mod init**
make[2]： *** [Makefile:84：/home/ubuntu/lede/build_dir/target-x86_64_musl/adguar
dhome-0.107.43/.built] **Error 1**
make[2]： Leaving directory '/home/ubuntu/lede/feeds/packages/net/adguardhome'
time:package/feeds/packages/adguardhome/compile#o.29#0.12#0.38
**ERROR**: package/feeds/packages/adguardhome failed to build.
make[1]： *** [package/Makefile:116：package/feeds/packages/adguardhome/compile]
Error 1
make[1]: Leaving directory '/home/ubuntu/lede'"
make： *** [/home/ubuntu/lede/include/toplevel.mk:231： package/feeds/packages/adg
uardhome/compile] **Error 2**
ubuntu@VM-16-14-ubuntu:~/lede
**请教kenzok8大佬和其他大神，为什么会出现此error？怎么解决这个error？**

fairok, 2024-01-29:
楼主的问题解决了么？

kenzok8, 2024-01-29:
应该是补丁的问题，应该已经修复了

fairok, 2024-01-29:
> 应该是补丁的问题，应该已经修复了

在2小时前，我用github云编译，还是出现和楼主一样的错误。

smdx, 2024-02-27:
关于在ledeのopenwrt下编译adguardhome的临时解决方法
**问题分析：**在编译adgardhome时，会生成AdGuardHome和adguardhome两个大小写不同的文件夹
```
sed -i '/^\t\$(call Build\/Prepare\/Default)/a \\tif [ -d "$(BUILD_DIR)\/AdGuardHome-$(PKG_VERSION)" ]; then \\\n\t\tmv "$(BUILD_DIR)\/AdGuardHome-$(PKG_VERSION)\/"* "$(BUILD_DIR)\/adguardhome-$(PKG_VERSION)\/"; \\\n\tfi' package/feeds/kenzo/adguardhome/Makefile
sed -i '/gzip -dc $(DL_DIR)\/$(FRONTEND_FILE) | $(HOST_TAR) -C $(PKG_BUILD_DIR)\/ $(TAR_OPTIONS)/a \\t( cd "$(BUILD_DIR)\/adguardhome-$(PKG_VERSION)"; go mod tidy )' package/feeds/kenzo/adguardhome/Makefile

```

出处
https://blog.suysker.xyz/archives/428

kenzok8, 2024-02-27:
我试着源码修改下吧

smdx, 2024-02-27:
> 我试着源码修改下吧

是不是可以做个补丁什么的？怕以后adg更新版本又出问题。谢谢

kenzok8, 2024-02-27:
嗯，可以的，先看看编译正常不

closed 2024-03-23 by kenzok8

mcusee, 2024-06-10:
rm -rf feeds/packages/net/adguardhome
请问解决没还是删除报错

make[3]: *** [Makefile:87: /home/eason/lede/build_dir/target-x86_64_musl/adguardhome-0.107.51/.built] Error 1
make[3]: Leaving directory '/home/eason/lede/feeds/kenzo/adguardhome'
time: package/feeds/kenzo/adguardhome/compile#2.72#4.73#23.79
ERROR: package/feeds/kenzo/adguardhome failed to build.
make[2]: *** [package/Makefile:116: package/feeds/kenzo/adguardhome/compile] Error 1
make[2]: Leaving directory '/home/eason/lede'
make[1]: *** [package/Makefile:110: /home/eason/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/eason/lede'
make: *** [/home/eason/lede/include/toplevel.mk:231：world] 错误 2