kenzok8/openwrt-packages #445: upx的makefile文件有问题

ChaingTsung, 2024-01-10:
编译时提示upx下载失败，但是手动在编译机上`wget -P dl https://github.com/upx/upx/releases/download/v4.2.1/upx-4.2.1-src.tar.xz `又可以成功下载，但是下载成功后再次编译还是不行，说upx下载失败

ChaingTsung, 2024-01-10:
SHELL= flock /home/yoki/op/tmp/.root-copy.flock -c 'cp -fpR /home/yoki/op/build_dir/target-x86_64_musl/binutils-2.41/.pkgdir/libctf/. /home/yoki/op/staging_dir/target-x86_64_musl/root-x86/'
touch /home/yoki/op/staging_dir/target-x86_64_musl/root-x86/stamp/.libctf_installed
echo "libctf" >> /home/yoki/op/staging_dir/target-x86_64_musl/pkginfo/binutils.default.install
mkdir -p /home/yoki/op/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/yoki/op/tmp/.root-copy.flock -c 'cp -fpR /home/yoki/op/build_dir/target-x86_64_musl/binutils-2.41/.pkgdir/libopcodes/. /home/yoki/op/staging_dir/target-x86_64_musl/root-x86/'
touch /home/yoki/op/staging_dir/target-x86_64_musl/root-x86/stamp/.libopcodes_installed
echo "libopcodes" >> /home/yoki/op/staging_dir/target-x86_64_musl/pkginfo/binutils.default.install
mkdir -p /home/yoki/op/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/yoki/op/tmp/.root-copy.flock -c 'cp -fpR /home/yoki/op/build_dir/target-x86_64_musl/binutils-2.41/.pkgdir/binutils/. /home/yoki/op/staging_dir/target-x86_64_musl/root-x86/'
touch /home/yoki/op/staging_dir/target-x86_64_musl/root-x86/stamp/.binutils_installed
echo "binutils" >> /home/yoki/op/staging_dir/target-x86_64_musl/pkginfo/binutils.default.install
mkdir -p /home/yoki/op/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/yoki/op/tmp/.root-copy.flock -c 'cp -fpR /home/yoki/op/build_dir/target-x86_64_musl/binutils-2.41/.pkgdir/objdump/. /home/yoki/op/staging_dir/target-x86_64_musl/root-x86/'
touch /home/yoki/op/staging_dir/target-x86_64_musl/root-x86/stamp/.objdump_installed
echo "objdump" >> /home/yoki/op/staging_dir/target-x86_64_musl/pkginfo/binutils.default.install
mkdir -p /home/yoki/op/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/yoki/op/tmp/.root-copy.flock -c 'cp -fpR /home/yoki/op/build_dir/target-x86_64_musl/binutils-2.41/.pkgdir/ar/. /home/yoki/op/staging_dir/target-x86_64_musl/root-x86/'
touch /home/yoki/op/staging_dir/target-x86_64_musl/root-x86/stamp/.ar_installed
echo "ar" >> /home/yoki/op/staging_dir/target-x86_64_musl/pkginfo/binutils.default.install
make[3]: Leaving directory '/home/yoki/op/package/devel/binutils'
time: package/devel/binutils/compile#0.57#0.30#0.72
make[3]: Entering directory '/home/yoki/op/feeds/kenzo/ucl'
make[3]: Leaving directory '/home/yoki/op/feeds/kenzo/ucl'
time: package/feeds/kenzo/ucl/host-compile#0.26#0.12#0.35
make[3]: Entering directory '/home/yoki/op/feeds/kenzo/upx'
UPX_UCLDIR=/home/yoki/op/staging_dir/host make -C /home/yoki/op/build_dir/hostpkg/upx-4.2.1/src LDFLAGS="-L/home/yoki/op/staging_dir/host/lib -L/home/yoki/op/staging_dir/hostpkg/lib -L/home/yoki/op/staging_dir/target-x86_64_musl/host/lib" CXX="/home/yoki/op/staging_dir/host/bin/g++"
make[4]: *** /home/yoki/op/build_dir/hostpkg/upx-4.2.1/src: No such file or directory.  Stop.
make[3]: *** [Makefile:61: /home/yoki/op/build_dir/hostpkg/upx-4.2.1/.built] Error 2
make[3]: Leaving directory '/home/yoki/op/feeds/kenzo/upx'
time: package/feeds/kenzo/upx/host-compile#0.18#0.07#0.22
    ERROR: package/feeds/kenzo/upx [host] failed to build.
make[2]: *** [package/Makefile:126: package/feeds/kenzo/upx/host/compile] Error 1
make[2]: Leaving directory '/home/yoki/op'
make[1]: *** [package/Makefile:122: /home/yoki/op/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/yoki/op'
make: *** [/home/yoki/op/include/toplevel.mk:232: world] Error 2

lirg-111, 2024-01-22:
upx

修改makefile
PKG_VERSION:=4.2.2
PKG_HASH:=42ee0455eea610ef7ee732aa1f657b34a351ebcfa64a24c1e2a7aaec74c1e038

ChaingTsung, 2024-02-10:
> UPX
> 
> 修改 makefile PKG_VERSION:=4.2.2 PKG_HASH:=42ee0455eea610ef7ee732aa1f65
7b34a351ebcfa64a24c1e2a7aaec74c1e038

还是不行
[image]

make -j1 V=s
实际上，源地址下载不对
```
...
time: package/devel/binutils/compile#0.45#0.38#0.81
make[3]: Entering directory '/home/yoki/op/feeds/kenzo/ucl'
make[3]: Leaving directory '/home/yoki/op/feeds/kenzo/ucl'
time: package/feeds/kenzo/ucl/host-compile#0.24#0.10#0.31
make[3]: Entering directory '/home/yoki/op/feeds/kenzo/upx'
mkdir -p /home/yoki/op/dl
SHELL= flock /home/yoki/op/tmp/.upx-4.2.2-src.tar.xz.flock -c '         /home/yoki/op/scripts/download.pl "/home/yoki/op/dl" "upx-4.2.2-src.tar.xz" "42ee0455eea610ef7ee732aa1f657b34a351ebcfa64a24c1e2a7aaec74c1e038" "" "https://github.com/upx/upx/releases/download/v4.2.2/upx-4.2.2-src.tar.xz"    '
+ curl -f --connect-timeout 20 --retry 5 --location https://github.com/upx/upx/releases/download/v4.2.2/upx-4.2.2-src.tar.xz/upx-4.2.2-src.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     9    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location https://sources.cdn.openwrt.org/upx-4.2.2-src.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0   146    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location https://sources.openwrt.org/upx-4.2.2-src.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0   146    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0
curl: (22) The requested URL returned error: 404
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location https://mirror2.openwrt.org/sources/upx-4.2.2-src.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0   153    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0
curl: (22) The requested URL returned error: 404
Download failed.
No more mirrors to try - giving up.
make[3]: *** [Makefile:62: /home/yoki/op/dl/upx-4.2.2-src.tar.xz] Error 2
make[3]: Leaving directory '/home/yoki/op/feeds/kenzo/upx'
time: package/feeds/kenzo/upx/host-compile#0.53#0.10#3.79
    ERROR: package/feeds/kenzo/upx [host] failed to build.
make[2]: *** [package/Makefile:126: package/feeds/kenzo/upx/host/compile] Error 1
make[2]: Leaving directory '/home/yoki/op'
make[1]: *** [package/Makefile:122: /home/yoki/op/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/yoki/op'
make: *** [/home/yoki/op/include/toplevel.mk:232: world] Error 2

```

lirg-111, 2024-02-11:
修改makefile
PKG_VERSION:=4.2.2
PKG_RELEASE:=1

PKG_SOURCE_PROTO:=git
PKG_SOURCE_URL:=https://github.com/upx/upx.git
PKG_SOURCE_VERSION:=099c3d829e80488af7395a4242b318877e980da4
PKG_MIRROR_HASH:=8bee3cd1ada2e4bfbe396348d2aecb3e044f92cdae2b7010838e36997fe6b3c3

lirg-111, 2024-02-11:
还有一种方法：
查找upx文件所在目录：
whereis upx
比如是在：/usr/bin/upx
拷贝文件到指定目录
cp /usr/bin/upx /home/***/lede/staging_dir/host/bin/
cp /usr/bin/upx-ucl /***/matt/lede/staging_dir/host/bin/
***替换成你自己的

closed 2024-02-11 by ChaingTsung