kenzok8/openwrt-packages #454: 编译出错package/feeds/packages/mosdns failed to build

woni928, 2024-02-28:
gcc" CXX="aarch64-openwrt-linux-musl-g++" CGO_CFLAGS="-Os -pipe -march=armv8-a+crypto -mtune=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/workdir/openwrt/build_dir/target-aarch64_generic_musl/mosdns-5.3.1=mosdns-5.3.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_CPPFLAGS="-I/workdir/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/usr/include -I/workdir/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/include/fortify -I/workdir/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/include" CGO_CXXFLAGS="-Os -pipe -march=armv8-a+crypto -mtune=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/workdir/openwrt/build_dir/target-aarch64_generic_musl/mosdns-5.3.1=mosdns-5.3.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_LDFLAGS="-L/workdir/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/usr/lib -L/workdir/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/lib -znow -zrelro" CGO_ENABLED=0 GOPATH="/workdir/openwrt/build_dir/target-aarch64_generic_musl/mosdns-5.3.1/.go_work/build" GOCACHE="/workdir/openwrt/tmp/go-build" GOMODCACHE="/workdir/openwrt/dl/go-mod-cache" GOENV=off GOTOOLCHAIN=local /usr/bin/env bash ../../lang/golang//golang-build.sh build -v -buildvcs=false -trimpath -ldflags "all=-buildid '1708791019' -linkmode external -extldflags '-L/workdir/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/usr/lib -L/workdir/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/lib -Wl,-z,now -Wl,-z,relro'" -ldflags " -X main.version=v5.3.1 -buildid '1708791019' -linkmode external -extldflags '-L/workdir/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/usr/lib -L/workdir/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/lib -Wl,-z,now -Wl,-z,relro'"
Finding targets

Building targets
-linkmode requires external (cgo) linking, but cgo is not enabled

make[3]: *** [Makefile:48: /workdir/openwrt/build_dir/target-aarch64_generic_musl/mosdns-5.3.1/.built] Error 1
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/net/mosdns'
time: package/feeds/packages/mosdns/compile#0.44#0.41#0.52
ERROR: package/feeds/packages/mosdns failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/mosdns/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'

woni928, 2024-02-28:
https://github.com/sbwml/luci-app-mosdns/issues/198

解决办法：
如果使用了 go1.22 ，mosdns 源码必须包含这个提交 ：https://github.com/sbwml/luci-app-mosdns/commit/664253d1e2beff1e3ca7134c69aed9abe116bf8c
也就是直接拉取最新代码即可。

kenzok8, 2024-02-28:
1、如果使用了 go1.22编译，你拉取上面的提交更新是一种方法，因为mosdns里有低版本的go补丁了，所以才会编译出错
2、如果本仓库的最新mosdns 而且配合go1.22编译，可以用 
sed -i '39i\GO_PKG_TARGET_VARS:=$(filter-out CGO_ENABLED=%,$(GO_PKG_TARGET_VARS)) CGO_ENABLED=1\n' mosdns/Makefile
sed -i '40i\\nGO_PKG_TARGET_VARS:=$(filter-out CGO_ENABLED=%,$(GO_PKG_TARGET_VARS)) CGO_ENABLED=1' v2dat/Makefile
去添加（或者手动去添加也行）
3、话说上游仓库更新了代码，去掉了GO_PKG_TARGET_VARS:=$(filter-out CGO_ENABLED=%,$(GO_PKG_TARGET_VARS)) CGO_ENABLED=1，为什么你要添加？

woni928, 2024-02-28:
> 1、如果使用了 go1.22编译，你拉取上面的提交更新是一种方法，因为mosdns里有低版本的go补丁了，所以才会编译出错 2、如果本仓库的最新mosdns 而且配合go1.22编译，可以用 sed -i '39i\GO_PKG_TARGET_VARS:=$(filter-out CGO_ENABLED=%,$(GO_PKG_TARGET_VARS)) CGO_ENABLED=1\n' [...]

那我闷了，既然上游又去除了，那就不需要这个了吧？

kenzok8, 2024-02-28:
本来就不需要了 如果你用go1.22编译就需要

woni928, 2024-02-29:
谢谢

closed 2024-02-29 by woni928

love-dy, 2024-02-29:
xiaorouji的passwall 需要go1.22

wuzhiyi51, 2024-03-31:
执行这个可以通过
$ export GOPROXY=https://goproxy.cn