kenzok8/openwrt-packages #456: 编译passwall出错

hztbhtbh, 2024-03-08:
我给X86编译L大的lede软路由固件，编译回退5.4的版本，在./scripts/feeds install -a出现如下错误，结果编译也失败了！
请各位大佬看看怎么修改：
`tbh@ubuntu20:~/lede$ ./scripts/feeds install -a

WARNING: Makefile 'package/feeds/small/shadowsocksr-libev/Makefile' has a dependency on 'libopenssl-legacy', which does not exist
WARNING: Makefile 'package/feeds/small/shadowsocksr-libev/Makefile' has a dependency on 'libopenssl-legacy', which does not exist
WARNING: Makefile 'package/feeds/small/shadowsocksr-libev/Makefile' has a dependency on 'libopenssl-legacy', which does not exist
WARNING: Makefile 'package/feeds/small/shadowsocksr-libev/Makefile' has a dependency on 'libopenssl-legacy', which does not exist
WARNING: Makefile 'package/feeds/small/shadowsocksr-libev/Makefile' has a dependency on 'libopenssl-legacy', which does not exist
Installing all packages from feed packages.
Installing all packages from feed luci.
Installing all packages from feed routing.
Installing all packages from feed telephony.
Installing all packages from feed kenzo.
Installing all packages from feed small.`

编译后出错如下：

`ede/tmp/stage-eudev/usr/share/pkgconfig
install -d -m0755 /home/tbh/lede/tmp/stage-eudev/usr/lib
cp -fpR /home/tbh/lede/build_dir/target-x86_64_musl/eudev-3.2.10/ipkg-install/lib/libudev.so* /home/tbh/lede/tmp/stage-eudev/usr/lib
cp -fpR /home/tbh/lede/build_dir/target-x86_64_musl/eudev-3.2.10/ipkg-install/lib/libudev.a /home/tbh/lede/tmp/stage-eudev/usr/lib
install -d -m0755 /home/tbh/lede/tmp/stage-eudev/usr/lib/pkgconfig
cp -fpR /home/tbh/lede/build_dir/target-x86_64_musl/eudev-3.2.10/ipkg-install/lib/pkgconfig/libudev.pc /home/tbh/lede/tmp/stage-eudev/usr/lib/pkgconfig
find /home/tbh/lede/tmp/stage-eudev -name '*.la' | xargs -r rm -f;
if [ -f /home/tbh/lede/staging_dir/target-x86_64_musl/packages/eudev.list ]; then /home/tbh/lede/scripts/clean-package.sh "/home/tbh/lede/staging_dir/target-x86_64_musl/packages/eudev.list" "/home/tbh/lede/staging_dir/target-x86_64_musl"; fi
if [ -d /home/tbh/lede/tmp/stage-eudev ]; then (cd /home/tbh/lede/tmp/stage-eudev; find ./ > /home/tbh/lede/tmp/stage-eudev.files);  SHELL= flock /home/tbh/lede/tmp/.staging-dir.flock -c ' mv /home/tbh/lede/tmp/stage-eudev.files /home/tbh/lede/staging_dir/target-x86_64_musl/packages/eudev.list && cp -fpR /home/tbh/lede/tmp/stage-eudev/* /home/tbh/lede/staging_dir/target-x86_64_musl/; '; fi
rm -rf /home/tbh/lede/tmp/stage-eudev
touch /home/tbh/lede/staging_dir/target-x86_64_musl/stamp/.eudev_installed
mkdir -p /home/tbh/lede/bin/targets/x86/64/packages /home/tbh/lede/build_dir/target-x86_64_musl/eudev-3.2.10/ipkg-x86_64/libeudev/CONTROL /home/tbh/lede/staging_dir/target-x86_64_musl/pkginfo
install -d -m0755 /home/tbh/lede/build_dir/target-x86_64_musl/eudev-3.2.10/ipkg-x86_64/libeudev/usr/lib
cp -fpR /home/tbh/lede/build_dir/target-x86_64_musl/eudev-3.2.10/ipkg-install/lib/libudev.so* /home/tbh/lede/build_dir/target-x86_64_musl/eudev-3.2.10/ipkg-x86_64/libeudev/usr/lib
find /home/tbh/lede/build_dir/target-x86_64_musl/eudev-3.2.10/ipkg-x86_64/libeudev -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
export CROSS="x86_64-openwrt-linux-musl-"  NO_RENAME=1 ; NM="x86_64-openwrt-linux-musl-nm" STRIP="/home/tbh/lede/staging_dir/host/bin/sstrip -z" STRIP_KMOD="/home/tbh/lede/scripts/strip-kmod.sh" PATCHELF="/home/tbh/lede/staging_dir/host/bin/patchelf" /home/tbh/lede/scripts/rstrip.sh /home/tbh/lede/build_dir/target-x86_64_musl/eudev-3.2.10/ipkg-x86_64/libeudev
rstrip.sh: /home/tbh/lede/build_dir/target-x86_64_musl/eudev-3.2.10/ipkg-x86_64/libeudev/usr/lib/libudev.so.1.6.3: shared object
(cd /home/tbh/lede/build_dir/target-x86_64_musl/eudev-3.2.10/ipkg-x86_64/libeudev/CONTROL; ( echo "$CONTROL"; printf "Description: "; echo "$DESCRIPTION" | sed -e 's,^[[:space:]]*, ,g'; ) > control; chmod 644 control; ( echo "#!/bin/sh"; echo "[ \"\${IPKG_NO_SCRIPT}\" = \"1\" ] && exit 0"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_postinst \$0 \$@"; ) > postinst; ( echo "#!/bin/sh"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_prerm \$0 \$@"; ) > prerm; chmod 0755 postinst prerm;  )
install -d -m0755 /home/tbh/lede/bin/packages/x86_64/base
/home/tbh/lede/staging_dir/host/bin/fakeroot /home/tbh/lede/staging_dir/host/bin/bash /home/tbh/lede/scripts/ipkg-build -m "" /home/tbh/lede/build_dir/target-x86_64_musl/eudev-3.2.10/ipkg-x86_64/libeudev /home/tbh/lede/bin/packages/x86_64/base
Packaged contents of /home/tbh/lede/build_dir/target-x86_64_musl/eudev-3.2.10/ipkg-x86_64/libeudev into /home/tbh/lede/bin/packages/x86_64/base/libeudev_3.2.10-1_x86_64.ipk
rm -rf /home/tbh/lede/build_dir/target-x86_64_musl/eudev-3.2.10/.pkgdir/libeudev.installed /home/tbh/lede/build_dir/target-x86_64_musl/eudev-3.2.10/.pkgdir/libeudev
mkdir -p /home/tbh/lede/build_dir/target-x86_64_musl/eudev-3.2.10/.pkgdir/libeudev
install -d -m0755 /home/tbh/lede/build_dir/target-x86_64_musl/eudev-3.2.10/.pkgdir/libeudev/usr/lib
cp -fpR /home/tbh/lede/build_dir/target-x86_64_musl/eudev-3.2.10/ipkg-install/lib/libudev.so* /home/tbh/lede/build_dir/target-x86_64_musl/eudev-3.2.10/.pkgdir/libeudev/usr/lib
touch /home/tbh/lede/build_dir/target-x86_64_musl/eudev-3.2.10/.pkgdir/libeudev.installed
mkdir -p /home/tbh/lede/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/tbh/lede/tmp/.root-copy.flock -c 'cp -fpR /home/tbh/lede/build_dir/target-x86_64_musl/eudev-3.2.10/.pkgdir/libeudev/. /home/tbh/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/tbh/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.libeudev_installed
echo "libeudev" >> /home/tbh/lede/staging_dir/target-x86_64_musl/pkginfo/eudev.default.install
make[3]: Leaving directory '/home/tbh/lede/package/qat/libs/eudev'
time: package/qat/libs/eudev/compile#22.02#5.47#26.05
make[2]: Leaving directory '/home/tbh/lede'
make[1]: *** [package/Makefile:110: /home/tbh/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/tbh/lede'
make: *** [/home/tbh/lede/include/toplevel.mk:231: world] Error 2
`
我根据这个地址：[https://github.com/fw876/helloworld/issues/1394](url)
修改后输入编译还是出错，看来修改的不对，应该是缺少libopenssl-legacy 依赖

`sed -i 's/ +libopenssl-legacy//g' feeds/small/shadowsocksr-libev/Makefile`

hztbhtbh, 2024-03-09:
已经有大佬解决，感谢大佬：
[https://github.com/xiaorouji/openwrt-passwall/discussions/2969](url)
复制上面地址游览器粘贴打开

closed 2024-03-10 by hztbhtbh

Bpc-CPU, 2024-08-05:
不用复制的-->[https://github.com/xiaorouji/openwrt-passwall/discussions/2969](https://github.com/xiaorouji/openwrt-passwall/discussions/2969)