kenzok8/openwrt-packages #460: X-ray每次都出错

magi-Z, 2024-03-16:
make[3]: *** [Makefile:53: /home/magi/lede/build_dir/target-aarch64_generic_musl/Xray-core-1.8.9/.built] Error 1
make[3]: Leaving directory '/home/magi/lede/feeds/small/xray-core'
time: package/feeds/small/xray-core/compile#0.22#0.47#2.23
    ERROR: package/feeds/small/xray-core failed to build.
make[2]: *** [package/Makefile:116: package/feeds/small/xray-core/compile] Error 1
make[2]: Leaving directory '/home/magi/lede'
make[1]: *** [package/Makefile:110: /home/magi/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/magi/lede'
make: *** [/home/magi/lede/include/toplevel.mk:231：world] 错误 2

dongfangy2011, 2024-03-16:
make[3]: *** [Makefile:44: /home/df/lede+/build_dir/target-x86_64_musl/v2ray-plugin-5.15.0/.built] Error 1
make[3]: Leaving directory '/home/df/lede+/feeds/smpackage/v2ray-plugin'
time: package/feeds/smpackage/v2ray-plugin/compile#0.15#0.13#0.26
    ERROR: package/feeds/smpackage/v2ray-plugin failed to build.
make[2]: *** [package/Makefile:116: package/feeds/smpackage/v2ray-plugin/compile] Error 1
make[2]: Leaving directory '/home/df/lede+'
make[1]: *** [package/Makefile:110: /home/df/lede+/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/df/lede+'
make: *** [/home/df/lede+/include/toplevel.mk:231：world] 错误 2

dongfangy2011, 2024-03-16:
同样的问题一直困扰着无法解决，等待中

kenzok8, 2024-03-16:
#148 帖子里有解决办法
_另外x-ray编译问题，尽量到small仓库提交 issue_

closed 2024-03-16 by kenzok8