kenzok8/openwrt-packages #482: adguardhome error

wangweitung, 2024-07-08:
```
go: ../../go.mod requires go >= 1.22.5 (running go 1.22.1; GOTOOLCHAIN=local)

Building targets
go: ../../go.mod requires go >= 1.22.5 (running go 1.22.1; GOTOOLCHAIN=local)

make[3]: *** [Makefile:84: /home/Disk/4Tsdd/lede/build_dir/target-x86_64_musl/AdGuardHome-0.107.52/.built] Error 1
make[3]: Leaving directory '/home/Disk/4Tsdd/lede/feeds/kenzo/adguardhome'
time: package/feeds/kenzo/adguardhome/compile#0.29#0.24#0.50
    ERROR: package/feeds/kenzo/adguardhome failed to build.
make[2]: *** [package/Makefile:116: package/feeds/kenzo/adguardhome/compile] Error 1
make[2]: Leaving directory '/home/Disk/4Tsdd/lede'
make[1]: *** [package/Makefile:110: /home/Disk/4Tsdd/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/Disk/4Tsdd/lede'
make: *** [/home/Disk/4Tsdd/lede/include/toplevel.mk:231: world] Error 2
```

kenzok8, 2024-07-08:
https://github.com/kenzok8/openwrt-packages/issues/480#issuecomment-2212394752[注意golang版本](url)

kenzok8, 2024-07-08:
#480

wangweitung, 2024-07-08:
> #480

thanks~

closed 2024-07-08 by wangweitung