kenzok8/openwrt-packages #485: 是op.dllkids.xyz的流量又满了吗？

Jackie264, 2024-07-17:
好多天都访问不了了

kenzok8, 2024-07-17:
是的，访问量不小

95486755, 2024-07-17:
这半个月不到 200G没了？

534002646, 2024-07-18:
> ![image](https://private-user-images.githubusercontent.com/39034242/349505162-8ed82196-dc40-46ff-88d4-2f7f00de7715.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjEyNzA1MDksIm5iZiI6MTcyMTI3MDIwOSwicGF0aCI6Ii8zOTAzNDI0Mi8zNDk1MDUxNjItOGVkODIxOTYtZGM0MC00NmZmLTg4ZDQtMmY3ZjAwZGU3NzE1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MTglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzE4VDAyMzY0OVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTRmZTUyNjk0ZTM0ZWUwY2FiYWYwODJjODgxMjM4YWY5NDQ4YWM5OGU2MTVmMDIwMzVkN2Q4OWYxYzRiZjljNDYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.6j8x3dNl1ofq6eLkRwNvTVDq7G7P6HC0792Uf35fBtQ) 是的，而且这vps不便宜 ![image](https://private-user-images.githubusercontent.com/39034242/349505964-e87c470d-e0db-4e57-9010-d20f0a9ca6df.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjEyNzA1MDksIm5iZiI6MTcyMTI3MDIwOSwicGF0aCI6Ii8zOTAzNDI0Mi8zNDk1MDU5NjQtZTg3YzQ3MGQtZTBkYi00ZTU3LTkwMTAtZDIwZjBhOWNhNmRmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MTglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzE4VDAyMzY0OVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTUyMzE0OWExODVlZTI5Y2Y4MjRlYzZhNWJlOGQ0ZWZmYTI3ZDBlZjlmNGM3NjBkYWRmYzQyOWE3NTU5ZWYxMzAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.OJLiznkmIpqMrkz2K-IQh6FyJCvwYz2LWOzueh9mwRM)

试试用cloudflare的workers+r2存储，流量免费的

qq1316237299, 2024-08-01:
我感觉好久都没用了

kenzok8, 2024-08-02:
这次到8月11日恢复

Jackie264, 2024-10-10:
> 这次到8月11日恢复

10月啥时候恢复:(