kenzok8/openwrt-packages #488: luci-app-argon异常，进后台报错，使用的是ImmortalWrt 23.05.3

tonghuabande, 2024-08-02:
之前使用一直都正常，今天编译安装之后就无法使用了
/usr/lib/lua/luci/ucodebridge.lua:23: /usr/lib/lua/luci/template.lua:181: Failed to execute template 'themes/argon/header'.
[image]

kenzok8, 2024-08-02:
https://github.com/kenzok8/openwrt-packages/issues/438#issuecomment-2236742458

kenzok8, 2024-08-02:
本来放两个18.06的argon没意义，所以一个是老揭力的23.05 另外一个我修改的argone 适配18.06的，群友说版本混乱，哎

tonghuabande, 2024-08-02:
> 本来放两个18.06的argon没意义，所以一个是老揭力的23.05 另外一个我修改的argone 适配18.06的，群友说版本混乱，哎

大佬，那我想使用23.05版本的应该如何修改呢

kenzok8, 2024-08-02:
1 删除错误主题

opkg info luci-theme*
opkg remove luci-theme-argon

2 openwrt主页系统下多余的无效主题名
nano /etc/config/luci

3 /usr/lib/lua/luci/view/themes/
目录下删除失效包

然后上传安装argon.ipk

tonghuabande, 2024-08-02:
> 1 删除错误主题
> 
> opkg info luci-theme* opkg remove luci-theme-argon
> 
> 2 openwrt主页系统下多余的无效主题名 nano /etc/config/luci
> 
> 3 /usr/lib/lua/luci/view/themes/ 目录下删除失效包
> 
> 然后上传安装argon.ipk
> 
> 用WinSCP等连接openwrt

如果想要编译进固件里应该怎么操作呢

tonghuabande, 2024-08-02:
> 我还是把luci-theme-argon改成默认的23.05吧，argone为18.06就行了

感谢大佬

closed 2024-08-19 by kenzok8