kenzok8/openwrt-packages #491: 编绎lede固件luci-app-alist错误

adwzlad, 2024-08-18:
不懂，随便截一段
mkdir -p /home/h/lede/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/h/lede/tmp/.root-copy.flock -c 'cp -fpR /home/h/lede/build_dir/target-x86_64_musl/linux-firmware-20240709/.pkgdir/bnx2x-firmware/. /home/h/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/h/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.bnx2x-firmware_installed
echo "bnx2x-firmware" >> /home/h/lede/staging_dir/target-x86_64_musl/pkginfo/linux-firmware.default.install
make[3]: Leaving directory '/home/h/lede/package/firmware/linux-firmware'
time: package/firmware/linux-firmware/compile#2.30#0.34#2.64
make[2]: Leaving directory '/home/h/lede'
make[1]: *** [package/Makefile:110: /home/h/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/h/lede'
make: *** [/home/h/lede/include/toplevel.mk:231：world] 错误 2

kenzok8, 2024-08-18:
[编译没问题哦](https://github.com/kenzok8/litte/releases)

adwzlad, 2024-08-19:
自定义luci,最后加入alist就编绎不成功

adwzlad, 2024-08-19:
> 自定义luci,最后加入alist就编绎不成功

make[1]: *** [package/Makefile:110: /home/h/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/h/lede'
make: *** [/home/h/lede/include/toplevel.mk:231：world] 错误 2

adwzlad, 2024-08-19:
不知道什么原因

adwzlad, 2024-08-19:
自定义luci，不要alist编绎成功，反之就这个报错信息

adwzlad, 2024-08-25:
windows11的锅
win11下，hyper-v虚拟硬盘形式或者外挂物理硬盘都是不能编绎成功.内置硬盘感觉有问题，U盘设备经常性掉设备。驱动什么的都正常安装的。
换win10，什么问题也没有了

adwzlad, 2024-08-25:
> windows11的锅
> win11下，hyper-v虚拟硬盘形式或者外挂物理硬盘都是不能编绎成功.内置硬盘感觉有问题，U盘设备经常性掉设备。驱动什么的都正常安装的。
> 换win10，什么问题也没有了

adwzlad, 2024-08-25:
搞的我一会怀疑PC硬件设备，一会怀疑系统，一会怀疑U盘设备。WIN11都装了好几个版本，N久之前，WIN11使用也没有问题的，WIN11现在肯定对不很久的设备也不友好

adwzlad, 2024-08-25:
工

closed 2024-09-06 by kenzok8