kenzok8/openwrt-packages #493: luci-app-adguardhome与adguardhome配置文件大小写不匹配

wang1zhen, 2024-09-05:
当前版本，编译openwrt后会在/etc目录下产生两个adguardhome.yaml配置文件（大小写不同）
[image]
其中，=AdGuardHome.yaml=是luci界面下手动设置的对象，同时也是实际生效的配置文件。但同时，=adguardhome.yaml=会在通过网页管理时生成，但实际使用时会被覆盖。

https://github.com/kenzok8/openwrt-packages/blob/61b3d9f8f28cee1e02cb69bb899e2d37f5ed760c/adguardhome/Makefile#L48

https://github.com/kenzok8/openwrt-packages/blob/61b3d9f8f28cee1e02cb69bb899e2d37f5ed760c/luci-app-adguardhome/Makefile#L27

可以看到，这两处Makefile间命名没有统一，因此导致了上述的问题。

鉴于openwrt自己的adguardhome包中也使用的是小写，能否将luci-app-adguardhome的配置文件，统一使用小写，来避免重复与混淆？

kenzok8, 2024-09-05:
收到，已去除luci的AdGuardHome.yaml

wang1zhen, 2024-09-06:
似乎编译后仍然存在`AdGuardHome.yaml`?

这是一个我刚编译完固件的内容：

[image]

我理解的逻辑应该是，通过web页面配置adguardhome的话，会生成`/etc/adguardhome.yaml`文件，但实际读取使用的是`/etc/AdguardHome.yaml`。另外这个默认的`/etc/AdguardHome.yaml`在同步了你的修改后依然存在，好像也比较奇怪？

smdx, 2024-09-20:
@kenzok8 
去除luci的AdGuardHome.yaml后遇到问题了

ImmortalWrt 21.02-SNAPSHOT r20111-fc9fe90e5a / LuCI openwrt-21.02 branch git-23.098.38725-847bd6b

基础设置：
[image]

手动设置：
/usr/lib/lua/luci/model/cbi/AdGuardHome/manual.lua:70: bad argument #1 to 'access' (string expected, got nil)
stack traceback:
	[C]: in function 'access'
	/usr/lib/lua/luci/model/cbi/AdGuardHome/manual.lua:70: in function 'func'
	/usr/lib/lua/luci/cbi.lua:66: in function 'load'
	/usr/lib/lua/luci/dispatcher.lua:1353: in function '_cbi'
	/usr/lib/lua/luci/dispatcher.lua:1024: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:479: in function </usr/lib/lua/luci/dispatcher.lua:478>

kenzok8, 2026-02-24:
自动巡检：该 Issue 长时间未更新（>=180 天），先关闭以便集中处理当前问题。

如问题仍存在，请按最新源码重新复现后提交新 Issue，并附：
- 固件版本与目标架构
- 完整编译/运行日志（从第一个报错开始）
- 复现步骤

Auto triage: this issue has been inactive for >=180 days, so it is being closed for backlog cleanup.
If the issue still exists, please open a new issue with updated logs and clear reproduction steps.