kenzok8/openwrt-packages #497: 请问R4S有固件，输入IP地址后，无法进入固件主页，跳出Error Unhandled exception during request dispatching /usr/lib/lua/luci/ucodebridge.lua:23: /usr/lib/lua/luci/template.lua:181: Failed to execute template 'themes/argone/header'. A runtime error occurred: /usr/lib/lua/luci/dispatcher.lua:68: attempt to index global '__entries' (a nil value)

netlovehf, 2024-09-16:
Error
Unhandled exception during request dispatching
/usr/lib/lua/luci/ucodebridge.lua:23: /usr/lib/lua/luci/template.lua:181: Failed to execute template 'themes/argone/header'.
A runtime error occurred: /usr/lib/lua/luci/dispatcher.lua:68: attempt to index global '__entries' (a nil value)
stack traceback:
/usr/lib/lua/luci/dispatcher.lua:68: in function 'node'
[string "/usr/lib/lua/luci/view/themes/argone/header..."]:20: in main chunk

In error(), file [C]
called from function run (/usr/lib/lua/luci/ucodebridge.lua:23)
called from function [anonymous function] (/usr/lib/lua/luci/ucodebridge.lua:37)
In [anonymous function](), file /usr/share/ucode/luci/runtime.uc, line 133, byte 10:
  called from function [arrow function] (/usr/share/ucode/luci/runtime.uc:141:63)
  called from function render ([C])
  called from function [anonymous function] (/usr/share/ucode/luci/runtime.uc:141:64)
  called from function [anonymous function] (/usr/share/ucode/luci/dispatcher.uc:941:44)
  called from anonymous function (/www/cgi-bin/luci:39:13)

 `            die(ex);`
  Near here --------^

netlovehf, 2024-09-16:
源码用的是immortalwrt的openwrt-23.05分支

kenzok8, 2024-09-16:
themes/argone 主题出错了，
仓库里luci-theme-argon 适配23.05 不适配18.06
argone 只适配18.06，就是L大的

netlovehf, 2024-09-17:
> themes/argone 主题出错了， 仓库里luci-theme-argon 适配23.05 不适配18.06 argone 只适配18.06，就是L大的 删除错误的主题
> 
> ```
> opkg info luci-theme*
> opkg remove luci-theme-argone
> ```
> 
> 然后下载argon再安装

请问这个操作是在编译固件之前吗？还是在编译好的固件中操作？我用的源码是immortalwrt的openwrt-23.05分支

kenzok8, 2026-02-24:
自动巡检：该 Issue 长时间未更新（>=180 天），先关闭以便集中处理当前问题。

如问题仍存在，请按最新源码重新复现后提交新 Issue，并附：
- 固件版本与目标架构
- 完整编译/运行日志（从第一个报错开始）
- 复现步骤

Auto triage: this issue has been inactive for >=180 days, so it is being closed for backlog cleanup.
If the issue still exists, please open a new issue with updated logs and clear reproduction steps.

closed 2026-02-24 by kenzok8